JMRI/JMRI #11248: XML files NOT displaying "LABEL" correctly

jimexler, 2022-07-28:
According to the instructions, in a variable, when a LABEL is present, the text contained in the LABEL will be listed instead of the ITEM text. This is not working.
In Decoder Pro 4.15+R37, the LABEL displayed correctly instead of the ITEM.
In Decoder Pro 4.26+R381, the ITEM text is displayed even when a LABEL is present and continues through V5
This makes the roster file nearly useless unless you can translate the generic numbering system of the variables. It has also affected some of the pane files causing the descriptions for the values to be the generic items.
The generic numbering system is important to make the system more universal but being able to use the LABEL function to customize this for individual decoders is equally valuable.
PLEASE review and fix the problem. Thanks

bobjacobsen, 2022-07-28:
I don't see this, but I may be looking in the wrong place.  Could you provide a decoder type and specific variable as an example?  Thanks!

jimexler, 2022-07-28:
[LocoRosterFiles.zip](https://github.com/JMRI/JMRI/files/9211138/LocoRosterFiles.zip)
Attached are two loco roster files, one from each of the referenced versions. They are for the same loco, just minor editing and thus the change in the file date. I also have a file from V5 and it shows the same problem.
Starting at LN 45, compare the text shown. 111520 shows FL..controls Headlight. 070522 shows FL...controls output 1. (no definition for output 1)
Headlight is the LABEL and output 1 is the ITEM.
Continue thru the file and you will see this is predominant throughout.
Thank you for looking at this. Let me know if I can be of any help.
Jim

bobjacobsen, 2022-07-28:
Thanks for the files.  I was looking at the GUI, which seems to be displaying the label properly.

So the problem you're seeing is that the XML file itself is listing contents by item instead of by label?  Can I ask why that's important to you? JMRI's file formats are considered internal, so that we can change them as needed, but we try not to unnecessarily break external uses.  Do you perhaps have some external process that's using the files?

jimexler, 2022-07-28:
I don't have any external use for the roster file other than reviewing the values. As I stated before, the generic numbering listing is of no use to me, I don't want to have to lookup each value. 
Why did this work correctly on DP 4.18 and not on 4.26?

bobjacobsen, 2022-07-28:
You review the values in the XML file directly rather than through DecoderPro itself?  I'm pretty sure nobody was considering that use case when the change was made.  

Could you review values by e.g. looking at the printable version DecoderPro makes?

The change was made in PR #9763 for JMRI 5.1.2.  You can see more info there and in Issue #9495, but basically the reason was that label names change (for example when a decoder definition is updated or when somebody is using a language other than US English), but that the item names don't change.  Using the item names means that the file is more flexible for DecoderPro itself.

jimexler, 2022-07-28:
I have DP on a laptop in my hobby room but sometimes look at the roster files on my desktop in my office. Looks like the DP printout contains the pertinent data but at 24 pages for my TSU2, I think I'll find another way.
Thanks for your knowledge and answers, I really appreciate what you do..I was hoping this was a simple setting but never thought of the multiple language issue.
Looks like it is best to close this issue; can you do that or do I?

bobjacobsen, 2022-07-28:
Thanks for the kind words.

You could share the files between those two files with e.g. DropBox

https://www.jmri.org/help/en/html/setup/Dropbox.shtml

It takes a little setup, but once done you can run DP on either machine to see the roster entries.

I’ll close the Issue.

closed 2022-07-28 by bobjacobsen