JMRI/JMRI #11253: Zimo MS-series sound decoders, new files

nigelcliffe, 2022-07-29:
Attached are various decoder files for the Zimo MS-series of sound decoders.  

They represent a first-stab at covering these new decoders.  Derived from an edit of the MX-series version 40 files.  

The files are: 
Zimo_MS_small_v4    - decoder file covering MS series decoders. 
Zimo ZIP file - contains various components to go into the Zimo sub-folder for decoders.   Two of the files replace current files (providing small bug fixes to those), the other seven are new files for the MS-series where there are significant departures from the MX-series, or where the version numbering of the two decoder families cause clashes.  

Still to do is a work on consisting, where there are some new Zimo features to add, and integrate with the existing consist panes.  

Tested briefly on one MS490 decoder.    These need feedback from users covering any features missed or errors.   
I think they are good enough for a test-release at this early stage of the JMRI development cycle. 

regards

- Nigel

[Zimo.zip](https://github.com/JMRI/JMRI/files/9221765/Zimo.zip)

[Zimo_MS_small_v4.zip](https://github.com/JMRI/JMRI/files/9221764/Zimo_MS_small_v4.zip)

bobjacobsen, 2022-08-04:
I'll commit these. Sorry for the delay.

bobjacobsen, 2022-08-04:
PR #11276

closed 2022-08-05 by danielb987