JMRI/JMRI #11265: Turnout based signal mast being updated by layout

arthkl, 2022-07-31:
Hello,

Apologies if this is not the correct way to discuss those topics. Please advise of a different way.
What I'm after is a way to intercept LnTurnout.handleReceivedOpSwAckReq() to update a signal mast that uses that turnout.
Background is a layout that is automated and PanelPro is used to visualize the layout. 

I couldn't find any support to easily identify a TurnoutSignalMast using a given turnout and my current solution - to go through all signals every time a turnout message has been received in order to find, via aspects, an assigned turnout - is clearly far from optimal.
Other changes are the TurnoutMastSignal.setAspect() would have to support - update only.
Meaning it would only update the aspect without sending the command back to the LocoNet. 

So my question: 
Has this ever been considered? Does this feature make sense from an overall solution point of view?
Please let me know. Also if I'm missing something and there is already a way to accomplish this scenario.

Cheers, Arco

danielb987, 2022-07-31:
@bobjacobsen 
Is this something you could answer? I have no idea.

devel-bobm, 2022-08-01:
@arthkl, please clarify the problem you have and the behavior you desire.  It is unclear to me exactly how many different problems you face, and it is unclear to me what any one of those challenges might be.

Be aware that JMRI's LocoNet developers have, in the past, attempted to differentiate between JMRI's Turnout control messages for controlling SE8C-based Turnout-controlled signal heads, and those LocoNet Turnout control messages issued to Turnouts associated with SE8C signal head which were sent by LocoNet agents other than JMRI.  Those attempts were fruitless, and the code had to be backed-out.  

Have you consiedered whether the "Bushby" feature (available with at least some LocoNet command stations) and JMRI's Bushby support can
 solve (any of) the problem(s) you face?

bobjacobsen, 2022-08-01:
I'm not sure what your asking for.  More detail would help.

As a general rule it is _not_ possible to recover the state of a SignalHead or SignalMast by listening to the messages that are sent to the layout.  Because of that, JMRI only drives SignalHeads and SignalMasts, it does not listen for changes that might have been done by something else. 

In limited cases, it might be possible to make some kind of listening work.  Various people have tried that.  They usually find a different way to do what they want to do, though.  So knowing more about what behavior you're trying to achieve would help.

arthkl, 2022-08-01:
@devel-bobm, thank you for reaching out. The layout is setup with turnout based signal masts. (TurnoutSignalMast).
On the panel I can use the associated SignalMastIcons to control the TurnoutSignalMasts. Basically changing the aspects on the panel does change the aspects of the physical signal on the layout. So far so good - 
however, the signals are not controlled via PanelPro. The layout is running in an automated mode where all signals are controlled by other components connected to the LocoNet (mainly Lissy - Uhlenbrock).
What I'm after is a way to change the SignalMastIcons on the panel based on the status of the linked Turnouts.

I 'played' with the various models and my "Proof of Concept" ,if you will, is to intercept the LocoNet messaged in the LnTurnout, go through all signals (very inefficient) to find a signal the uses that turnout and to change the aspect of the signal based in the the status of the turnout, without sending the command back to the LocoNet (otherwise one would end up with an endless loop)
Essentially I want to use PanelPro to visualize the entire Layout without necessarily controlling the layout.

bobjacobsen, 2022-08-01:
Do you already have the SignalMast completely set up?  Or is this something you're still working on creating?

Do you need to be able to change the signals from the panel?  If that's optional, there's a simpler way to set this up.  But it's different from using Turnout-based masts, hence my first questions.

arthkl, 2022-08-01:
@bobjacobsen Thank you for your answer. our messages seem to have crossed in the cloud ;-)
According to your answer the idea to change TurnoutSignalMask aspects via LocoNet messages, that are generated outside of JMRI, is not part of the solution.
Controlling the signals via JMRI is the supported method.

Yes, I have the signal mast configured and can control the physical signals on the layout via the panel.
That part is working fine.

It's the other way around I was trying to get working. 
Well, I have this working after changing LnTurnout and TurnoutSignalMast slightly. 
The signals on my panel changing their aspects based on other events happening on my layout without me touching any of them.
However, this has only a character of a POC as I'm concerned about the efficiency of that solution.

bobjacobsen, 2022-08-01:
Efficiency is nice, of course, but a modern machine can do a _lot_ of work in the time that it takes to send or receive a LocoNet message.  

You could perhaps save a bit of time by replacing the search with a pre-made map.  Python dictionaries are good for this and very, very fast.

KenC57, 2022-08-01:
My thoughts along this ask what are the boards that:
1. drive the signals
2. are the signals
3. what sort of signal rule set are they displaying

From these we might see other ways to tell what's going on.

Next for your displays to work, the masts at least should be virtual masts so they won't send anything back to
the layout. I'm thinking a script or LogixNG where each one watches what impacts a mast and computes the
displayed aspect from that. Each part only has to listen to the turnout or other messages that impact that
mast.

Another thing is some of the hardware may have other messages that would help to know what the aspect may be
on the layout. Last a question, is it critical that the panel reproduce exactly what is on the layout? You
might find that a simpler set of aspects is easier to compute than representing exactly what a mast is
showing.

devel-bobm, 2022-08-01:
I would recommend a solution which relies upon Jython.  The "dictionary" which @bobjacobsen recommends is a good place to start - define how certain turnouts relate to a JMRI Mast as appropriate.  Define the JMRI Masts as "virtual" masts so that they are entirely de-coupled from hardware.  

Provide a Turnout Listener which monitors for your externally-generated Turnout control messages.  Use those messages (and your knowledge of how the signaling hardware interprets those messages) to determine the equivalent "aspect" for the appropriate virtual mast.  Update the JMRI mast based on that aspect.

This solution requires that you do the necessary coding, because only you will know the relationship between turnouts and masts, and because only you will know the relationship between the requested turnout states and the effect on the mast.  Note that this last bit can be difficult because turnout-controlled masts (usually) will transition from a valid state, thru one or more "intermediate" states which may or may not correspond to a "valid" Aspect for the mast, before receiving the last turnout state which achieves the final "valid" Aspect.  Exactly how to handle those intermediate states is not necessarily straight-forward...

(It would be _far_ easier if you were working with "aspect-controlled" signal masts, rather than anything that requires multiple LocoNet messages to define a single mast aspect...)

arthkl, 2022-08-01:
Doesn't the panel configuration defines the relationship between signal mast and turnout?
```
  <signalmasts class="jmri.managers.configurexml.DefaultSignalMastManagerXml">
    <turnoutsignalmast class="jmri.implementation.configurexml.TurnoutSignalMastXml">
      <systemName>IF$tsm:DB-HV-1969:block($0001)</systemName>
      <userName>Loop1</userName>
      <unlit allowed="no" />
      <aspect defines="Hp0">
        <turnout>LT101</turnout>
        <turnoutstate>thrown</turnoutstate>
      </aspect>
      <aspect defines="Hp1">
        <turnout>LT101</turnout>
        <turnoutstate>closed</turnoutstate>
      </aspect>
    </turnoutsignalmast>
```
My hope was to ask the signal manager for a signal that relies on a certain turnout - at least that's what my current POC is based on, anyway.

To answer @KenC57 's question. The layout is based on Uhlenbrock technology. IntelliBoxII is driving the signals via - we call it turnouts here. Depending on the complexity of a signal one or two turnouts are necessary. (Simple Block works with one, Entry and Departure signals usually require two turnouts)

devel-bobm, 2022-08-01:
As @bobjacobsen wrote in his first repy, 
> As a general rule it is not possible to recover the state of a SignalHead or SignalMast by listening to the messages that are sent to the layout. Because of that, JMRI only drives SignalHeads and SignalMasts, it does not listen for changes that might have been done by something else.

JMRI developers have, in the past, tried to implement mechanisms to "reverse-engineer" signal head and signal mast "aspect" information from control message information.  For simple signaling, it can be possible, but for more-complex signaling and more-complex control mechanisms, it can become incredibly difficult.  

I would note that JMRI developers once tried to provide the sort of functionality you seem to be asking for, specifically for Digitrax SE8C signal head control via LocoNet Turnout messages.  They "failed miserably", and had to back-out their changes and remove the attempted feature.

JMRI's design philosophy is something like "start from the general, and refine towards the specific".  It can be incredibly challenging to start from the specific and work towards the general.  And turned out to be problematic for the SE8C case mentioned above.

Your case is probably simple-enough for a self-made custom solution, and I would recommend that approach, using the suggestions already mentioned in this issue.

bobjacobsen, 2022-08-01:
> Doesn't the panel configuration defines the relationship between signal mast and turnout?

SignalMasts are 1st-class objects; they all act the same, hiding any internal structure, regardless of type.  So they do know about the Turnouts or other objects they use, but they don’t make that available in any interface.

You could add that to your local copy, of course.  Then a script could index over all the signal masts and make a look-up dictionary of the results.

arthkl, 2022-08-01:
Thanks, yes I'm doing something similar in my local copy. Although I'm doing this right in the LNTurnout class.

I basically wanted to get an idea if the concept of updating turnout based signal masts based on LocoNet messages is something the JMRI community was already entertaining. Because this is done with regular turnouts, I thought it could be.

Like to thank you for all the input. 
I'll maintain those changes in my local copy.

KenC57, 2022-08-01:
Arthkl,

Your statement "Simple Block works with one, Entry and Departure signals usually
require two turnouts" points to a simple enough bit of coding. I'd start coding
for those block signals. Then try the entry/departure signals, you may find
there is enough of a pattern to make it work. But it may be easy enough to do
this in Logix or LogixNG.

arthkl, 2022-08-01:
KenC57,

Thank you your advice.
As of now it seems I have a pretty good handle on those signals that are controlled by two turnouts.
It gets more complicated when an entry signal has a distance signal that is dynamically linked to different departure signals.
Going to have to deal with 4 different turnouts that in combination represent one aspect.
Basically one string that needs to be passed into the setAspect() function. Essentially following this table.
[image]

At this point I'm not sure how this table would translate into this configuration screen:
[image]

I haven't looked into Logix or LogixNG, yet. 
Just started with JMRI two weeks ago - lot to read  ;-)

Cheers, Arco

KenC57, 2022-08-02:
Arco,

That table is good. But the key is the masts on the panel are virtual, not
turnout masts. The 'Mast Driver' is virtual. You use Logix to set the aspects.

For that mast, you have one Logix that has a conditional statement for each line
in the table.

First,  condition, if turnout 1 thrown, action set mast aspect Hp0.
Second, condition, if turnout 1 closed and turnout 2 closed and turnout 3 thrown
and turnout 4 thrown, action set mast aspect Hp1+Vr0.
Third, condition, if turnout 1 closed and turnout 2 closed and turnout 3 thrown
and turnout 4 closed, action set mast aspect Hp1+Vr0.

Etc... keep going down the table. If two are setting the same aspect, see if you
can remove one of the turnouts from the conditions.
Note: 2nd and 3rd could be just testing turnouts 1, 2 and 3 since the state of
turnout 4 doesn't make a difference, you set Hp1+Vr0.
Same note about the 6th and 7th lines, turnout 4 has no impact as both of those
are setting Hp2+Vr0.

Now that would mean a whole new Logix for the next mast and repeat. But the
aspect of one mast could also be one of the conditions for another mast.

Now it may seem like a lot of setting up, but the nice thing is one Logix
controls one mast. The debugging can be simpler from that.

-Ken Cameron, Member JMRI Dev Team
www.jmri.org
www.fingerlakeslivesteamers.org
www.cnymod.com
www.syracusemodelrr.org

dsand47, 2022-08-02:
Ken,

For this kind of logic LogixNG might work better since it can do nested if-then-else. 

Dave Sand

KenC57, 2022-08-02:
Dave S,

Can LogixNG create a function? That would allow a function that takes the
arguments of which turnouts and mast and saves a lot of duplicated work. That
would work if one function for each table that is different. I've not had the
time to explore LogixNG yet.
-ken c

dsand47, 2022-08-02:
Ken,

Yes.

The basic process has the following components.
- A spreadsheet is created with the data, exported as a CSV file and loaded as a LogixNG table.
- A LogixNG conditional is created for each item which is triggered by some event(s).
- The conditional calls a LogixNG Module passing the key for the table row(s).
- The Module uses the table information and related JMRI information to perform its actions.

I use this process for a 12 track passenger facility to set the 11 turnouts based on a sensor going active for each track.  The sensors are in a JMRI Sensor Group so only the last one is active.

Dave Sand

arthkl, 2022-08-02:
Thank you Dave S and Ken,

I start to understand what Logix is about. It seems to be a powerful tool for various automation tasks. 
Not sure though if I want to build the signals with it.  My layout looks like this at the moment:
[image]

All signals have been build with TurnoutSignalMasts. 
I can control the signals on the layout from the Panel if needed and the signal icons on the Panel get updated if a signal gets changed on the layout. (the latter happens outside of JMRI - typically via Lissy Automatic functions). 
It basically works the same way as it does for regular turnouts.
I feel if I use virtualSignalMasts I loose the ability to control a signal from the Panel.

Nonetheless, I'll dive a little deeper into Logix and LogixNG to get a better understanding.

Thanks again.

dsand47, 2022-08-02:
Arco,

Where you able to figure out how to map the turnouts to the signal mast definition?

It takes between 4 and 7 turnouts to map the 7 aspects.  When an aspect changes, JMRI sends ONE turnout command.  Your table implies that 4 commands would be needed.

Here is an example using 4 turnouts.

Dave Sand

Basically one string that needs to be passed into the setAspect() function. Essentially following this table. Screen Shot 2022-08-01 at 6 34 01 PM <https://user-images.githubusercontent.com/63615014/182256791-453ce36d-d585-4594-b8b7-97ad755ba0d8.png>

At this point I'm not sure how this table would translate into this configuration screen: Screen Shot 2022-08-01 at 6 18 36 PM <https://user-images.githubusercontent.com/63615014/182255201-4f765fb0-d2ee-4ce4-be31-1448f664570b.png>

dsand47, 2022-08-02:
> Dave Sand,
>
> I did - for a brief moment ;-) - had a theory to look at those turnouts as a bitfield where JMRI - able send one turnout command at the time - changes one bit every time an aspect changes. Obviously that doesn't work when starting from the top again. So no, I have no idea how to handle this and might have to settle with changing the main aspects only.
>
> Since the majority of the aspect changes occur on my layout this would be less on an issue for me.
> What I really need to make sure that all turnouts contributing to a turnoutSignalMast are know to that signal so my updateTurnoutSignal function (a function I've added to the LnTurnout class) is able to find the signal with any given turnout.

Arco,

Here is a python fragment that gets the turnouts and their states for a turnout controlled signal mast.

        appearanceMap = self.mast.getAppearanceMap()
        for aspectName in appearanceMap.getAspects():
            turnoutName = self.mast.getTurnoutName(aspectName)
            turnoutState = self.mast.getTurnoutState(aspectName)

I tried to create a script that would show the aspect based on turnout states.  The problem is that scripting, along with Logix and LogixNG, are property change event driven.  In my sample configuration, LT101/Thrown sets Hp0.  When that aspect is to be set, the turnout command is sent to the LocoNet, but the turnout state has not changed, therefore no property state change.

Dave Sand

arthkl, 2022-08-04:
Dave Sand,

Yep, I'm doing something similar in the updateTurnoutSignalMast function. That part is actually working. What I'm struggling with is setting a turnout signal on the layout. (This is not too important as I have other means to set a signal - still curious how that should work)

Assuming a departure signal is configured like this:
[image]

Clicking through the aspects on the panel works perfect for the first cycle. The signal on the layout displays the aspects as desired.
However, when switching back to Hp00 and because JMRI only sends one turnout message at the time, nothing is changing the turnout 125. 125 remains closed.
Therefore, while the panel displays Hp00 - the layout is still showing Hp0+Sh1

It gets even more complicated if that signal mast has a distance signal.

Cheers, Arco

dsand47, 2022-08-04:
Arco,

What does the signal mast edit page look like?

Dave Sand

Assuming a departure signal is configured like this: Screen Shot 2022-08-04 at 8 42 34 AM <https://user-images.githubusercontent.com/63615014/182849854-d0201660-e9cc-414a-af10-9218e63e9732.png>

dsand47, 2022-08-04:
> Dave Sand,
>
> Looks like this (turnout numbers slightly different though)
>
> Screen Shot 2022-08-04 at 10 10 14 AM <https://user-images.githubusercontent.com/63615014/182868445-7ce6398a-e778-4466-a7d5-7af7d6e0da7c.png>

Arco,

I configured a test mast as you show.  When I click on the mast icon, it cycles throw the aspects as shown in the signal mast table and the appropriate turnout messages appear on the the LocoNet monitor.

I see that Hp00 and Hp1 are sharing a turnout.  Would that make a difference.

I don't understand your spreadsheet grid.  It does not seem to have any relationship to the JMRI configuration.

Maybe you should look at using the "Output Matrix Mast" driver.
The checked boxes are set to closed when the aspect is active.  The number of turnout commands depends on the difference between the previous aspect and the new aspect.

Apparently 4 is the minimum number so I created a dummy turnout.
 
I am using JMRI 5.1.2 for this test.

Dave Sand

arthkl, 2022-08-04:
Thank you Dave Sand,

Yes, it appears the "Output Matrix Mast" is the better choice. Perhaps the word "turnout" tripped me up.
With "Output Matrix Mast" I can mimic the desired matrix. 
[image]

Panel and Layout are sync now.

KenC57, 2022-08-04:
From the tables of turnouts you have shown, you would need to create something
that takes multiple turnouts for each aspect. The table view of things fit well
here. Think a set of columns. First column is the aspect to display. Second
through N (3 or 4?) would take the turnout and which way. I think that's what
you would need.

The output matrix seems to fit that, except that sometimes you only need one
output for an aspect, other times two or three. I don't know if you can leave a
few blank.

arthkl, 2022-08-04:
Thank you Ken and Dave Sand,
 Happy to report that I've changed the entire panel to use MatrixSignalMasts. 
Since you can specify the number of outputs when you create a MatrixSignalMast (1 for block, 2 for entry, 3 for exit and 4 for entry with distant) I was able to configure every combination needed without the need to configure dummy outputs.
Pretty cool concept actually.

The layout is now responding to every change on the panel and with my change to LnTurnout the panel is responding correctly to any change that happens on the layout. The system works both ways in the intended way.

Thanks again for your advice, MatrixSignalMast is the solution.

Cheers, Arco

closed 2022-10-04 by stale[bot]