JMRI/JMRI #11267: Getting a PR4 to work reliably as a simple programmer

spd4train, 2022-08-01:
Issue Type: __Bug__

I have a new PR4.  So far I am NOT thrilled.
I want to use it as a simple programmer on a separate programming track.  It is connected properly as best I can tell. (comm4, etc)  JMRI Decoder pro software is driving presently.  When I do a simple read of the basic information sheet I get no response messages from or PR4 or 306 time out errors.  The loco and its decoder were just working, responding to commands from a DCS50 on the same track (moved alligator clips from one pair or wires to the other).  Rramp meter is reading 16.5V DCC on track.
I should note that Digitrax's own software will NOT read a CV either.
At times I have been able to read and even write CVs using Decoder Pro, but NOT consistently.

DecoderPro 5.0+Rc441642522
[profile.zip](https://github.com/JMRI/JMRI/files/9229998/profile.zip)
[session.log](https://github.com/JMRI/JMRI/files/9229999/session.log)
[systemInfo.txt](https://github.com/JMRI/JMRI/files/9230000/systemInfo.txt)

OpenJDK 64-Bit Server VM Azul Systems, Inc. 11.0.15+10-LTS mixed mode
Windows 10 10.0 amd64

devel-bobm, 2022-08-01:
If you cannot read a loco CV via Digitrax software and the PR4, then it is almost impossible that JMRI will be able to read it.

I suggest that you focus on getting the PR4 to read reliably when using the Digitrax software, and only then work with JMRI.

FYI - this sort of problem is normally addressed via that JMRIUsers forum on groups.io, or, perhaps more appropriately in your case, the Digitrax-Users forum on groups.io.

spd4train, 2022-08-01:
Hmm, that is interesting. Programing track has one pair of alligator clips that connect to wires from either the PR4 or the DCS50. I have at Digitrax non-suggestion been using an 18V power supply when the unit supplied didn’t seem to cut the mustard.  I am working with two slightly different variations of LokSound 5 DCC decoders, one with capacitors and one without. OK, just went to other computer and train shed to see what was up.  RRampmeter definitely reading voltage as DCC (around 17V).  Pulled USB connection, dropped to nothing (black blank RRmeter).  Restarted everything.  Brought Soundproject up and RRmeter immediately read voltage.  Took Soundproject to CV read mode and status indicated that connection paused.  Put in a couple of CVs to read and off course it read them.  Checked all CVs and it obediently read them all.  (of course the program seems to have no way of saving the results to a file one can look at slowly or check against anything else).  So tried another decoder and back to response.  Restarted everything and still zilch response.  Haven’t gone back to first decoder yet. Set up has worked at least once. Read, complete with motor wiggle which I could observe, complete CVs for Loksound 5 DCC into DecoderPro.  Of course, something is wrong somewhere as that decoder no longer drives the motor or responds to commands.  Not sure if I wrote anything after that. Sent from Mail for Windows From: BreezlysSent: Monday, August 1, 2022 2:00 PMTo: JMRI/JMRICc: spd4train; AuthorSubject: Re: [JMRI/JMRI] Getting a PR4 to work reliably as a simple programmer (Issue #11267) The PR4 only sends a burst to the programming track when it's sending a command, and that burst is much too short for the rrampmeter to detect. Something's wrong if it's showing a constant 16.5V on your programming track. Either your PR4 is shot, or you're backfeeding the DCS50 into it (which will also ruin it).

devel-bobm, 2022-08-01:
I'd recommend that you ask on the Digitrax-Users list about "neutral mode" and its effect on non-Digitrax mobile decoders.

closed 2022-09-21 by bobjacobsen