JMRI/JMRI #11270: Uhlenbrock #63410 variables

silverailscolo, 2022-08-01:
I recently discovered that the v1 definition incorrectly stores combined variable in some CVS on the LocoNet Switch module 63410. The manual assigns specific digits instead of bits, like:
[image]
As the rightmost option has 4 or sometimes even more options, this won't fit in a bit. In other vars the single digit is on the left. 
I am busy updating this definition using a new variable type to mask digits instead of bits, both for DecVarValue as for EnumVarValue.

In the meantime users please be advised to wait for this update before configuring this item in DecoderPro.

danielb987, 2022-08-02:
Should this go to the JMRI users list too?

silverailscolo, 2022-08-02:
Will send it there too. Thanks for reminding

silverailscolo, 2022-08-02:
Should it be possible to code this with a radix mask?

danielb987, 2022-08-02:
> Should it be possible to code this with a radix mask?

I don't know. Maybe ask on the JMRI developers list?

silverailscolo, 2022-08-02:
After I wrote basic code to handle the above case, I came across a very similar expression in symbolicprog, so I think a radix mask should work.
In a 2019 JMRI-developers thread @bobjacobsen  mentioned:
```
partA = CVvalue%maskInt
partB = CVvalue/maskInt%maxValue
partC = CVvalue%(maskInt*(maxValue+1))

update/use partB as needed, and then reassemble (addition) after modifying partB as desired
```
Will add a short explanation with some examples to decoder help because I never grasped the radix style. Probably because I had no application for it 😀

silverailscolo, 2022-08-04:
Fixed in PR #11281. Closing issue

closed 2022-08-04 by silverailscolo