JMRI/JMRI #11285: Throttle causes crash

DPNelso, 2022-08-06:
I am having two issues that I can not find an answer for. I am working with an apple system with JMRI version 5.0 and Decoder Pro 3.4 and the NCE  Power Cab system. I am doing everything on a programming track.

Issue # 1 when I go into Decoder Pro Preference Connection and click “additional connector settings” and then SAVE  this will work for that session but will not be permanently saved.

Issue # 2  When I work with any diesel engine the following occurs.  When I put my locomotive on the programing track and start Decoder Pro my roster comes up. Then when I highlite an engine and click on the throttle button the throttle appears with the engine number and this is followed by the following  message page and the Decoder Pro application freezes.

This application will quit
java.lang.StackoverflowError

at imri.imrit.throttle.ThrottleFrame.loadThrottle(ThrottleFrame.lava:260
atjmm.jmnit.throttle.Throttlerrame.loadDefaultThrottle(ThrottleFrame.lava:229.
at imri.imrit.throttle.ThrottleFrame.loadThrottle(ThrottleFrame.java:288)

If I do not click the Throttle button I can continue to do everything else with the program and it works just fine.

danielb987, 2022-08-07:
@DPNelso 
Welcome to the JMRI project!
What computer do you have? Which model?
Can you copy the system console. Menu `Help -> System Console...`. Click on the button `Copy to clipboard` and then paste it here.

@dsand47 
Is this something you could look at?

DPNelso, 2022-08-07:
I am working with an iMac running High Sierra  10.13.6 which I have dedicated to my model railroad.

This is a copy of the System Console:
[image]

Hope this is what is needed.  Anything else please let me know.

Thank you

dsand47, 2022-08-07:
The text from the system console works better than a screenshot since a screenshot cannot be expanded without getting fuzzy.  Use the "Copy to clipboard" button.

The problem is the USB connection to the layout.  Which NCE command station are you using?  What USB device are you using?

Dave Sand

This is a copy of the System Console: Screen Shot 2022-08-07 at 8 24 57 AM <https://user-images.githubusercontent.com/110742396/183295959-ed57f44a-ca22-4600-9e73-7af4c17a1bf9.png>

Lionel642, 2022-08-07:
Hi,

Unfortunately the throttle issue was detected right after 5.0 release and
is fixed in later ones.

Regards,
Lionel

DPNelso, 2022-08-07:
In answer to Dave’s question I am using the NCE POWERCAB (version 1.65) and NCE USB interface for the powercab.

If I upgrade to JMRI version 5.1would that fix may throttle issue for now?

Attached is the system console copy.

Again thanks for your help and patience.

danielb987, 2022-08-07:
@DPNelso 
In order to attach a file to an issue, you need to use the web browser. Email attachments doesn't work for GitHub issues.

dsand47, 2022-08-07:
I don't know much about the NCE hardware.  Someone else will need to help.

Look at https://www.jmri.org/help/en/html/hardware/nce/NCE.shtml especially the section on trouble shooting.

For macOS USB issues, look at https://www.jmri.org/install/MacOSX.shtml#debug

Dave Sand

danielb987, 2022-08-07:
@DPNelso 
It might be better to ask on the JMRI users list. There are people that know the NCE hardware.
https://groups.io/g/jmriusers/topics
https://groups.io/g/jmriusers/join

dheap, 2022-08-08:
Installing NCE USB on Mac
===================
- Plug the USB side of the NCE USB into your Mac (you do not need to plug it into the PCP to do the next few steps.

- Go to System Report from Apple Menu->About This Mac->More Info.

- Look in the USB section and see if a Silicon Labs device is present. If not you have a USB hardware fault; cable, connector, port or device. That will need resolving before you go any further.

- Download and install SiLabs CP210x drivers from <https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers>.

- Unplug the NCE USB.

- Open a Terminal Window (from /Applications/Utilities), copy and paste the following command in EXACTLY (including spaces and punctuation) then press Enter.

while : ;do clear;ls -lt /dev|head;i=$((i+1));echo $i;sleep 1;done

- This command should show about 10 lines of output and then sit looping, with a counter incrementing every second.

- Plug your device back in and wait a few seconds. Two new items with names like cu.SLAB_USBtoUART and tty.SLAB_USBtoUART should appear in the list. They should disappear if you unplug your device. If this does not happen, you do not have the correct drivers installed for your device and you will need to solve that before proceeding further.

(With later macOS versions, you'll probably need to go to System Preferences->Security & Privacy->General at this stage and allow the new Silabs driver to load the first time.)

- Once you have found your device, write down the name you see.You want the cu.xxx version not the tty.xxx version.

- If you have got this far you know your device is be being seen by the Mac, that it has a matching driver installed and the name of the connection port "cu.xxx" you need to select in JMRI.

The best beginner-level/diagnostic settings for a Power Cab (any version) and an NCE USB (any version)  are:
- All USB Jumpers off (or at least 2, 3 & 4. Jumper 1 is ignored).
- JMRI preferences are:
  - System manufacturer: NCE
  - System connection:  NCE USB
  - Serial port: (The COM port number you saw appearing/disappearing)
  - USB Version: V7.x.x
  - System: PowerCab
  - Additional Connection Settings checkbox ON
    - Baud rate: 9600.

The JMRI system console log should then report V6.3.0 or V7.3.0.
If you don't get a version number in the log or the system does not work, Use Help->System Console...->Copy to clipboard and paste into a post for us to see.

Also please observe and advise the behaviour of the two LEDs (marked USB and CAB BUS) on the NCE USB (normal behaviour is both to flicker briefly when processing commands, either or both stuck on denotes a problem). Also the behaviour of the LED on the Power Cab Panel (normal behaviour is on, with a brief flicker when JMRI switches it into or out of programming mode).

closed 2022-10-08 by stale[bot]