JMRI/JMRI #11291: Rotated BlockContentsIcon do require a predefined size

arthkl, 2022-08-08:
This is related to #11208

The current implementation requires the size to be specified for BlockContentsIcon that are displayed with a rotation.
#11208 does create an icon for BlockContentsIcon that are not displayed horizontally (rotation deg=0)

However, the order in view the individual attributes (such as _text, _icon ..) are set in the MemoryIcon.displayState(Object key) destroys the icon almost immediately and cause the size to collapse to its minimum values. Consequently not displaying the label.

a smal change in the order of
original:
                } else if (val instanceof jmri.IdTag){
                    // most IdTags are Reportable objects, so
                    // this needs to be before Reportable
                    _icon = false;
                    _text = true;
                    setText(((jmri.IdTag)val).getDisplayName());
                    setIcon(null);
new:
                } else if (val instanceof jmri.IdTag){
                    // most IdTags are Reportable objects, so
                    // this needs to be before Reportable
                    _icon = false;
                    _text = true;
                    setText(((jmri.IdTag)val).getDisplayName());

danielb987, 2022-08-11:
@arthkl 
Do you have a panel file demonstrating this issue? I have created PR #11305 with your fix, but I'm not sure on how to test it.

arthkl, 2022-08-11:
Hi @danielb987,

[image]

The BlockContentsIcons on the left (270 deg angle) and on the right (90 deg angle) were not visible prior to that change.

I've attached the panel file.  (as txt)
[Moorbach.xml.txt](https://github.com/JMRI/JMRI/files/9308180/Moorbach.xml.txt)

arthkl, 2022-08-11:
@danielb987 
Just wanted to post out that the fix only addresses cases where the key passed into displayState(Object key) is an instance of IdTag. I haven't had time to test other cases like Sting, Reportable but would imagine that the order needs to be changed for those instances too.

closed 2022-10-11 by stale[bot]