JMRI/JMRI #11316: Change order of Digitrax system connections in Preferences

PUlvestad, 2022-08-15:
Several times a year we get users that are having trouble with their LocoBuffer-USB because they selected LocoBuffer instead. These 2 options aren't together in the list.
Can we have the selection of option sorted alphabetically so all the LocoBuffer options are together?

bobjacobsen, 2022-08-15:
They originally were.  Then there was a suggestion that devices that were current should go to the top.  Given that you can’t get a LocoBuffer or LocoBuffer-II these days, that made some sense and was done.

Should we switch back to alphabetical? Or is there some other order that’s better?

Bob

> On Aug 15, 2022, at 12:05 PM, Peter Ulvestad ***@***.***> wrote:
> 
> Several times a year we get users that are having trouble with their LocoBuffer-USB because they selected LocoBuffer instead. These 2 options aren't together in the list.
> Can we have the selection of option sorted alphabetically so all the LocoBuffer options are together?

—
Bob Jacobsen
***@***.***

bobjacobsen, 2022-08-15:
[image]

PUlvestad, 2022-08-15:
That list doesn't appear to be what is current at the top or the PR2 and 3 would be lower.
I think the issue is that the list doesn't always open at the top but rather the middle then people scroll down and grab what looks right and don't scroll up. 
My preference would be alphabetical but that's just one opinion.

bobjacobsen, 2022-08-15:
Happy to update it (hasn't been reordered in about 7 years), but I'm not sure alphabetical is best.  That would put "BT Locobridge" at the top; there aren't very many of those. And it would put "LocoBuffer" and "LocoBuffer-II" _before_ "LocoBuffer-NG" and "LocoBuffer-USB"

The order is manual in the code, so we can make it anything we want.

dsand47, 2022-08-15:
> That list doesn't appear to be what is current at the top or the PR2 and 3 would be lower.
> I think the issue is that the list doesn't always open at the top but rather the middle then people scroll [...]

It appears that you are looking at 4.26 or earlier.

At 5.0, the preferences combo boxes now use JComboBoxUtil to set the max rows to the screen height instead 8.

This change reduces hidden items.

PUlvestad, 2022-08-15:
Dave, 
That could be where I remember it opening in the middle. When I set up my profiles on this computer I'm positive I needed to  scroll up, I'm currently 5.13 and them see all. 
The current user with this possible issue is on 4.26

Bob
With Dave's comment maybe we should just leave it be.

bobjacobsen, 2022-08-15:
Maybe move the PR2 and PR3 down below the server choices?

PUlvestad, 2022-08-15:
That wouldn't hurt, especially the PR2

bobjacobsen, 2022-08-15:
PR #11317

danielb987, 2022-08-15:
How about a comment on the old items? Something like:
```
LocoNet LocoBuffer-NG
LocoNet LocoBuffer-USB
...
LocoNet LocoBuffer-II (Deprecated)
LocoNet LocoBuffer (Deprecated)
```

Where the word `Deprecated` should be changed to something better, like `Not for sale` or `Not available` or something.

PUlvestad, 2022-08-15:
Obsolete?

That would definitely describes the MS100 and PR2

danielb987, 2022-08-15:
> Obsolete?

That was the word I was looking for.

devel-bobm, 2022-08-15:
So far as I am concerned, there is a class of users for whom _it doesn't matter_ what order you choose, the user will _still_ choose an inappropriate setting.

There are _still_ users of the "obsolete" MS100.  And there are _still_ users of the "obsolete" PR2.  The idea that a JMRI developer considers them "obsolete" is irrelevant.

Consider that Digitrax has "retired":
- the MS100
- the PR2
- the PR3
- the PR3Xtra
- the DB100/DB100a with LA-2 and DT200 as "command station"
- the DB150
- the DCS100/DCS200
- the DCS50
- the DCS51
- the DCS240
- the DCS210.

Should JMRI list them _all_ as "Obsolete"?  Probably not.

I think the declaration by JMRI of _any_ hardware, whether Digitrax or non-Digitax, as "Obsolete" _is a slippery slope_...

PUlvestad, 2022-08-15:
You are correct, superseded or no longer in production does not mean obsolete.
Warnings where appropriate on the website hardware pages is sufficient.

KenC57, 2022-08-15:
I think the polite industry term for older hardware is 'Legacy' but that might
confuse some people.

geowar1, 2022-08-15:
There’s also “retro”… ;-)

Sent from GeoWar’s iPad Pro

closed 2022-08-16 by danielb987