JMRI/JMRI #11320: LoadAndStoreTest fail on Headless

danielb987, 2022-08-16:
@bobjacobsen @dsand47 
Lately there have been lots of failures on Headless CI on several branches. Any idea on what might cause this?
I don't think I have seen this error on Windows CI.

What I don't understand is why `scripts:PowerOff.py` has be changed to `preference:PowerOff.py`.

```
Error:  Failures: 
Error:    LoadAndStoreTest.loadAndStoreTest:37->LoadAndStoreTestBase.loadLoadStoreFileCheck:341->LoadAndStoreTestBase.checkFile:248 expected:<...uteScriptFile name="[scripts]:PowerOff.py" />> but was:<...uteScriptFile name="[preference]:PowerOff.py" />>
```
```
2022-08-16T16:07:16.8258763Z ERROR - match failed in LoadAndStoreTest: [main] jmri.configurexml.LoadAndStoreTestBase.checkFile()
2022-08-16T16:07:16.8262938Z ERROR -     file1:line 110: "      <routeScriptFile name="scripts:PowerOff.py" />" [main] jmri.configurexml.LoadAndStoreTestBase.checkFile()
2022-08-16T16:07:16.8267159Z ERROR -     file2:line 110: "      <routeScriptFile name="preference:PowerOff.py" />" [main] jmri.configurexml.LoadAndStoreTestBase.checkFile()
2022-08-16T16:07:16.8271574Z ERROR -   comparing file1:"/home/runner/work/JMRI/JMRI/java/test/jmri/implementation/configurexml/loadref/RouteTestExamples.xml" [main] jmri.configurexml.LoadAndStoreTestBase.checkFile()
2022-08-16T16:07:16.8276291Z ERROR -          to file2:"/home/runner/work/JMRI/JMRI/temp/temp/RouteTestExamples.xml" [main] jmri.configurexml.LoadAndStoreTestBase.checkFile()
2022-08-16T16:07:18.2120527Z [ERROR] Tests run: 8, Failures: 1, Errors: 0, Skipped: 0, Time elapsed: 2.748 s <<< FAILURE! - in jmri.implementation.configurexml.LoadAndStoreTest
2022-08-16T16:07:18.2121092Z [ERROR] loadAndStoreTest{File, boolean}[4]  Time elapsed: 0.331 s  <<< FAILURE!
2022-08-16T16:07:18.2326249Z org.junit.ComparisonFailure: expected:<...uteScriptFile name="[scripts]:PowerOff.py" />> but was:<...uteScriptFile name="[preference]:PowerOff.py" />>
2022-08-16T16:07:18.2326744Z 	at org.junit.Assert.assertEquals(Assert.java:117)
2022-08-16T16:07:18.2327072Z 	at org.junit.Assert.assertEquals(Assert.java:146)
2022-08-16T16:07:18.2327485Z 	at jmri.configurexml.LoadAndStoreTestBase.checkFile(LoadAndStoreTestBase.java:248)
2022-08-16T16:07:18.2328000Z 	at jmri.configurexml.LoadAndStoreTestBase.loadLoadStoreFileCheck(LoadAndStoreTestBase.java:341)
2022-08-16T16:07:18.2328584Z 	at jmri.implementation.configurexml.LoadAndStoreTest.loadAndStoreTest(LoadAndStoreTest.java:37)
```

danielb987, 2022-08-16:
A good example is this:
https://github.com/JMRI/JMRI/actions/runs/2867913734

All four runs on Headless CI have failed with this exact error.

bobjacobsen, 2022-08-16:
I've got no idea why that would be happening, particularly only on the headless tests (which run on Ubuntu linux).  I don't see that on my Mac, and apparently Windows is not seeing it in CI.  

I think we need to interest a Linux user to `git bisect` on this and see where it's coming from.

bobjacobsen, 2022-08-17:
That profile is used by the application-level (Cucumber?) tests that Paul Bender added.  There are a number of those with individual profiles. Not sure if those are accessed by other tests.

icklesteve, 2022-08-17:
I'm running local CI on a PR which runs all of the load / store tests in a temporary profile directory.

icklesteve, 2022-08-18:
Headless CI runs now seem to be passing fine.

closed 2022-08-18 by icklesteve