JMRI/JMRI #11334: JMRI Stops working when I plug in a second USB Serial Device

ptmahoney, 2022-08-22:
I just run out of ideas where to look on this one. I have a Arduino-Uno/Motor-shield DDC+ set up on USB Com5. Works great. Got a layout panel with 6 turnouts and can click each to throw the turnout. This is powered from a 12V supplied not using the USB as a power source. The servos are powered from a separate 5V DC supply.

I have an Arduino-Mega/Sensor-shield set up on USB Com7. This is powered from the 5V DC source. The USB is not providing power. As soon as I plug it in I find the turnouts won't switch. Unplug Com7 and the turnouts work again.

Any pointers where to look? The laptop has more then one USB outlet. So I tested with using the hub. Same problem. It's a Win10 PC. Any config settings I need to check?

devel-bobm, 2022-08-22:
Sharing a single power supply between two different device types is often problematic.  That common 5VDC supply is _likely_ the cause.  Try a separate 5VDC supply for the Mega, and see if the problem disappears.  If the problem disappears, then you cannot share that supply between the two different Arduino device types.

ptmahoney, 2022-08-22:
The Arduinos are on separate supplies. One is on a 12V supply. The other a 5V Supply. But I do wonder if I should connect them to a common ground.

ptmahoney, 2022-08-22:
Testing shows that creating a common ground between the 2 power supplies the kills the DCC signal output. When I plug in the 2 USB connectors I was creating a common ground. Looks like I need to rethink how I power the Arduinos.

devel-bobm, 2022-08-22:
Not enough information to give advice.  

Does JMRI give any error messages about your connections in the Console Log?  If so, please add any such messages to the issue.  

I would _guess_ that you have "unexpected current flow" from the Uno, via the DCC++ signal coming out of the "motor shield", somehow back thru the sensor board, and to the Mega's "GND".  And that current could be "crowbaring" the Mega's local power supply circuit.  Fixing that _will likely_ require re-work of your power supplies.

Adding a commoning wire between the power supplies might "help".  Or it might not.  It could cause one power supply or the other to shut down.  Lots of possibilities.  But such problems are in the realm of hardware, not JMRI.

I would suggest seeking help from a DCC++ forum for the hardware-related concerns.

geowar1, 2022-08-23:
Just a thought but perhaps plugging in the 2nd USB device renumbers the ports… IOW: Is Comm5 still Com5? Or is the 2nd device now Comm5 and the 1st one now Comm7? I've had Windows renumber ports in an inconsistent manner before to the point that I'd have to completely reboot a machine to get a consistent arrangement. Linux land is slightly better in that it names it's USB devices using their serial numbers (if one's available… which unfortunately isn't always the case).

ptmahoney, 2022-08-23:
It is definitely something to do with grounding between the PSU's. The Com Port stay as 5 and 7. Definitely a H/W issue and not JMRI. :)

devel-bobm, 2022-08-23:
Per @ptmahoney, his issue is a hardware one, not JMRI, so closing the issue.

closed 2022-08-23 by devel-bobm