JMRI/JMRI #11372: Operations day of week

spark-fuel-support, 2022-09-01:
I am enjoying the Operations features.  I wish that besides being able to set the time, the date could be modified.  With that, the train schedules should indicate the "current" day of week, and thus I could review how spur schedules work out for different days of the week.

danielb987, 2022-09-01:
@DanielBoudreau 
Could you take a look at this?

DanielBoudreau, 2022-09-01:
The day of the week is modified using the Train's Schedule tool.  The radio button shows which day is active, and the day is confirmed on the "Trains" window in the upper left hand corner.  Having the day changed based on a fast clock IMO could make things very complicated when building and terminating trains in real time.  Today OperationsPro doesn't access or use the fast clock.

spark-fuel-support, 2022-09-01:
Thanks Daniel.
  Thanks for trying to educate me.
  I had not pressed the activate button in the train schedules window, so
the day of week was not showing up on the trains page. That is a help.

  I think having the day of the week added to the date printed on the
various reports would be nice, but it would require a bit more
sophistication to combine the modeled year and date to assign the day of
week.

 I would like to let the trains run, and see where my planned layout
bottlenecks (run out of cars, run out of yard space, etc.  Is there any
plan for that?

Modeling Algoma Central Railway, November 1975.

DanielBoudreau, 2022-09-01:
To add the “Day of the Week” to Manifests and switch lists, see “Show Train Schedule” below

https://www.jmri.org/help/en/package/jmri/jmrit/operations/Operations.shtml#PrintOptions

You don’t have to use days, you can create other schedule names, like “Morning Run”, “Midday Run”, “Weekend”, etc.

To see how well cars move, you would simply Build and Terminate all of your trains several times.  The buttons on the bottom of the Trains and Train Schedule pages allow you to quickly do this.  If you’re using the random method for moving cars, most likely never find an issue.  If you’re using lots of custom loads and schedules, along with C/I tracks, that can lead to bottlenecks, car shortages and overloads.  Most don’t think it through enough.

Dan

spark-fuel-support, 2022-09-09:
Most things are making sense.
Some things are puzzling me.
1:  I have a spur for cleaning up cars- boxcars and gondolas get a
status of 'Dirty' after delivering certain loads, and this works well.
Even cabooses can be cleaned up.  But my engines can't be sent for cleanup,
refueling, etc.  Any suggestions?
 2:  Is there a way to show incremental time changes when using the Move
instruction?  The times are calculated on the various reports, but I can
only see original departure schedule time in the Trains view.

closed 2022-09-16 by DanielBoudreau