JMRI/JMRI #11381: Copied Panel xml file is not always usable on some Windows destination computers

Cliff-Anderson, 2022-09-03:
A Panel xml file created on a computer with a large screen and/or with multiple screens can be at risk.  
   
If the destination computer has either a smaller screen or has fewer screens, it is possible to have defined one or more panels with upper left corner location set outside of the physical screen pixel area.  
   
Apparently Linux and Mac operating systems provide corrective methods, but I have no experience with either OS.  
   
Windows fails to render any panel with upper left corner outside the screen area and provides no clues.  
   
Using a text editor to move the x and y settings back into the screen pixel area might not work if setting information in the "user-interface.xml" file overrides the values in the loaded panel file.  
   
See the lengthy discussion in the thread ["Layout Panels will not display"](https://groups.io/g/jmriusers/topic/93328449#207722) for context and suggested workaround details.  
   
The uploaded example panel file from that thread named "Howard 9.5.xml" is attached here as a zip file.  
   
[Howard 9.5.xml.zip](https://github.com/JMRI/JMRI/files/9481679/Howard.9.5.xml.zip)
   
One example of a heavy-handed workaround Config Profile folder named "Invisible_Panels_Second_Try.jmri" is also attached as a zip file.  
   
[Invisible_Panels_Second_Try.jmri.zip](https://github.com/JMRI/JMRI/files/9481691/Invisible_Panels_Second_Try.jmri.zip)

danielb987, 2022-09-04:
See PR #11384.

Cliff-Anderson, 2022-10-19:
A script named "UnHide Panels.py" has been created to allow a Windows user to see a description and decide to continue or quit.  
   
If the user choses to continue, a second dialog offers the option of one-by-one decisions to move any JmriJFrame to the upper left region of the main screen AKA \Display0 or just skip to the next Frame that is determined to be off the edge.  All Frames that are determined to be completely within the combined screen areas are ignored.  
   
The attached "UnHiding Panels.jmri" Config Profile will Start with the Loading the Panels and execute the script.  
   
The script in the jython subfolder will provide a fix for the "completely Off Screen" Panels.  

The method for determining the right edge of the rightmost screen has only been tested on a laptop using the HDMI connection to one TV screen and has not been verified to work with any other multiple screen hardware.

In order to test that script with multiscreen hardware configurations, more than once, it is necessary to Quit PanelPro and then either modify the file named "settings:\UnHiding_Panels.jmri\profile\4fb9c250-7bd3-11e8-8000-f81654d51042\user-interface.xml" to remove the entries between the lines:  
   
    <windowDetails xmlns="http://jmri.org/xml/schema/auxiliary-configuration/window-details-4-3-5.xsd">  
           and
    </windowDetails>
   
or else just delete that file and then restart PanelPro.  
   
Maybe this will help resolve the ambiguity of the "opt-out" discussion reported in PR https://github.com/JMRI/JMRI/pull/11384.
   
[UnHiding_Panels.jmri.zip](https://github.com/JMRI/JMRI/files/9817179/UnHiding_Panels.jmri.zip)

bobjacobsen, 2022-10-19:
Thanks!  I've created PR #11527 to move the script into the releases.  I removed the space from the name out of an excess of caution.

closed 2022-12-19 by stale[bot]