JMRI/JMRI #11386: Digitrax SE74, DS74, and PM74

joepopham, 2022-09-04:
Is there any timeframe currently when these Digitrax products s may be supported in a JMRI production or test software package?

PUlvestad, 2022-09-12:
I'm not a developer but the simple answer is no.

Digitrax doesn't readily supply the needed information to add new products. The developers must rely on the manuals, the Digitrax Knowledge Base **AND** reverse engineering. The developers are also all volunteers, they sacrifice family time and their own hobby time to work on JMRI so real life may affect when they can do programming.

joepopham, 2022-09-12:
Thanks.  Appreciate the reply.

closed 2022-11-13 by stale[bot]

bobjacobsen, 2023-04-13:
The SE74 and DS74 are in JMRI 5.3.5, out last week.  The PM74 will be in the next test release, in about a month.