JMRI/JMRI #11400: Unable to open NCE edit consist menu item.

dheap, 2022-09-10:
Unable to open NCE edit consist menu item. Worked in 5.1.3.

[messages.log.zip](https://github.com/JMRI/JMRI/files/9539740/messages.log.zip)

dheap, 2022-09-10:
As a current head of master.

silverailscolo, 2022-09-10:
I see a problem at the end  ```[AWT-EventQueue-0]
java.util.MissingResourceException: Resource 'MenuTools' not found
	at jmri.Bundle.retry(Bundle.java:205)
	at jmri.Bundle.handleGetMessage(Bundle.java:151)
	at jmri.jmrix.Bundle.retry(Bundle.java:108)
	at jmri.Bundle.handleGetMessage(Bundle.java:151)
	at jmri.jmrix.nce.Bundle.retry(Bundle.java:94)
	at jmri.Bundle.handleGetMessage(Bundle.java:151)
	at jmri.jmrix.nce.consist.Bundle.retry(Bundle.java:94)
	at jmri.Bundle.handleGetMessage(Bundle.java:151)
	at jmri.Bundle.handleGetMessage(Bundle.java:130)
	at jmri.jmrix.nce.consist.Bundle.getMessage(Bundle.java:42)
	at jmri.jmrix.nce.consist.NceConsistEditPanel.getMenus(NceConsistEditPanel.java:249)
	at jmri.util.swing.sdi.JmriJFrameInterface.show(JmriJFrameInterface.java:50)
```
Can only look at it tomorrow night.

dheap, 2022-09-11:
#11406 fixes the failure to open.

There may be other issues  with the tool but my  health isn't up to further debugging, testing 😕