JMRI/JMRI #11453: Valid XML fails for decoder-4-15.2.xsd

AFromAu, 2022-09-24:
Issue Type: __Bug__

Create a new CV file.

File header is:
```
<?xml version="1.0" encoding="utf-8"?>
<?xml-stylesheet type="text/xsl" href="../XSLT/decoder.xsl"?>
<!--Example CV file -->
<variables
	xmlns:xi="http://www.w3.org/2001/XInclude"
	xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
	xsi:noNamespaceSchemaLocation="http://jmri.org/xml/schema/decoder-4-15.2.xsd"
>
</variables>

```

Calling "Validate XML..." on this file results in the error:

> Error on line 1 of document http://jmri.org/xml/schema/decoder-4-15.2.xsd: White spaces are required between publicId and systemId.

That file is supplied with the latest version of DecoderPro, and documented as deprecating the previous schema.

DecoderPro 5.0+Rc441642522
Java HotSpot(TM) 64-Bit Server VM Oracle Corporation 11.0.2+9-LTS mixed mode
Windows 10 10.0 amd64

danielb987, 2022-09-24:
Can you upload the entire decoder file?

dsand47, 2022-09-24:
Change `4-15.2.xsd` to `4-15-2.xsd`.  Replace period with dash.

AFromAu, 2022-09-25:
> Change `4-15.2.xsd` to `4-15-2.xsd`. Replace period with dash.

Yes, that was the issue.  Thanks.  Error message is somewhat misleading.

closed 2022-09-25 by AFromAu