JMRI/JMRI #11456: Improve location documentation for "Creating A Custom Decoder File"

AFromAu, 2022-09-25:
[https://www.jmri.org/help/en/html/apps/DecoderPro/CreateDecoder.shtml](url)

> This new file should go in the decoders subdirectory in the JMRI User Files Location so that the program can find it.

The user directory has no existing "decoders" directory.  What is not obvious from the documentation is that:

- new files need to be created in <user_location.jmri>/xml/decoders
- any references to these files should be referred to by `/xml/decoders/...` or `http://jmri.org/xml/decoders/...` in order to be found by the "Validate XML..." function.  If using the short version the leading slash is required.

Proposed fixes:

> If the directory `xml/decoders` does not exist in your user location, create it.  Any new decoder files should go in this directory (or sub-directories) so that the program can find it.

And under "Include fragment files":

- Change all paths from `xml/decoders/...` to `/xml/decoders/...`.  Or use the full `http://jmri.org/xml/decoders/...` prefix if that is more correct.
- Possibly add section on "including your own fragment files".

danielb987, 2022-09-25:
@dsand47 @bobjacobsen 
Is this any of you wants to take a look at? I don't know how it should be.

AFromAu, 2022-09-26:
Related: [according to Bob Jacobson](https://groups.io/g/jmriusers/message/208599), one only needs to rebuild the decoder index when a new decoder file is added (or updated?), not when fragments are added.  This is also not clear from the documentation.

AFromAu, 2022-09-26:
Possibly related: If a manufacturer exists in the main database, and I add a new decoder file with the same manufacturer, I end up with 2 entries for that manufacturer (with different contents).

Example:

- File https://www.jmri.org/xml/decoders/MDE_HDD.xml exists and references "MD Electronics".
- I create local decoder file that references "MD Electronics".
- I regenerate decoder index and restart
- In DecoderPro, I go "+" to add a new loco
    - Decoder selection list appears

Expected behaviour:

- My entries for MD Electronics are aggregated into the existing entries for MD Electronics under a single folder.

Actual behaviour:

- There are two entries for "MD Electronics" - the original one from the supplied database, and a new one at the very bottom that contains my custom decoder entries.

bobjacobsen, 2022-11-10:
This one is still on my list to work on.

closed 2023-01-10 by stale[bot]