JMRI/JMRI #11466: Uhlenbrock DAISY Il-Digital-Zentrale support with JMRI

johnmrichey, 2022-09-27:
When using the USB interface 63120 connected to the 65120 Zentrale which command station type should be used?  I have tried both Intellibox-I and Intellibox-II but I can seem to be able to read a decoder when adding to my roster.  I have tried other listed types but they don't seem to work at all.

Thanks

danielb987, 2022-09-27:
@johnmrichey 
Welcome to the JMRI project!

Have you asked on the [JMRI users list](https://groups.io/g/jmriusers/topics)? There are more people there that might be able to help you. You can [join](https://groups.io/g/jmriusers/join) it or [browse](https://groups.io/g/jmriusers/topics) it.

johnmrichey, 2022-09-27:
I have not, but have now subscribed and awaiting approval.  Thanks for the tip.

silverailscolo, 2022-10-01:
The Uhlenbrock product number of your Zentrale would be 65210, I assume? If so, I can add a definition for LNCV configuration.

johnmrichey, 2022-10-01:
Yes, please that is the one.

closed 2022-12-04 by stale[bot]