JMRI/JMRI #11469: Documentation stale or incorrect for defaultItem

AFromAu, 2022-09-28:
https://www.jmri.org/help/en/html/apps/DecoderPro/CreateDecoderAdvanced.shtml

Section "Include/Exclude During Variable Expansion" provides examples of the use of element `defaultItem` which is contained in element `variable`.

However, `decoder-config.dtd` (as of 5.0) does not define `defaultItem`.  According to the DTD, default values can only be set as part of attribute `default`.

This creates issues when the default value varies based on `include` or `exclude` values.

bobjacobsen, 2022-09-28:
`decoder-config.dtd` is not the current specification for decoder contents.  There's a decoder-something schema in xml/schema that is (the one with the most current version number after it).  Sorry to not check myself, working from my phone.

AFromAu, 2022-09-28:
Ah, thanks.  I was misled because https://www.jmri.org/help/en/html/apps/DecoderPro/Debug.shtml "Debugging problems with XML files" refers to `decoder-config.dtd` not the replacement `.xsd` files.

**Change request**: update reference from `decoder-config.dtd` to newer `.xsd` file.

Issue with my XML was that `defaultItem` is required to go before the `enum` definition and I had put it after.

**Change request**: on https://www.jmri.org/help/en/html/apps/DecoderPro/CreateDecoderAdvanced.shtml
add the line

> Note: \<defaultItem\> must be the first element inside the \<variable\> element body.

somewhere after

> You can also use the include mechanism to specify the default value of a variable.

Either immediately or after the example.

(update: have sample change for 2nd request. Will link pull request.)

PR #11589 (2022-11-13): Issue 11469: improved decoder creation doc
Closes #11469
closed 2022-11-13 by danielb987