JMRI/JMRI #11488: Purple Light for USS CTC Panel

Mark-W-L, 2022-10-01:
resources/icons/USS/sensors does not have a purple light. Most PRR CTC panels use a blue light for Traffic indications, but some used purple. 
Mike Burgett at https://groups.io/g/CMRI-Users/message/9383 says blue.
https://position-light.blogspot.com blog from Sunday November 14 2021 references a Railway Signalling Journal article
https://jonroma.net/media/signaling/railway-signaling/1944/Pennsylvania%20Installs%20CTC%20On%2053%20Miles%20of%20Busy%20Single%20Track.pdf
This says that the Brady CTC panel had traffic lamps that were purple, although one page further on it refers to a blue traffic lamp.

For anyone who wishes to use a purple lamp gifs are attached. 

[image]
[image]

icklesteve, 2022-10-02:
Hi @Mark-W-L , If you could advise directory / preferred filenames I'll create a PR for inclusion ( assuming images created by yourself ) ?

Thanks,
Steve.

Mark-W-L, 2022-10-02:
Steve,
I created the images by taking the existing blue on and blue off lamps and changing the color using Photoshop elements.  They should go in resources/icons/USS/sensor as purple-on.gif and purple-off.gif, which will match the names of the existing lamps.
Thanks,
Mark

PR #11493 (2022-10-05): add Purple Light for USS CTC Panel
fixes #11488
closed 2022-10-05 by icklesteve