JMRI/JMRI #11509: NPE for .equals() in Block.java

SvenRosvall, 2022-10-11:
I am using Java 5.0. (The code issue below seems to exist in the latest master branch too.
I did not have the same problems in 4.26.

I get a lot of NPE like this:
`
java.lang.NullPointerException: Cannot invoke "Object.equals(Object)" because the return value of "jmri.Block.getValue()" is null
	at jmri.Block.goingActive(Block.java:1052)
	at jmri.Block.handleSensorChange(Block.java:895)
...
`
The code on line 1052 in Block.java is

                           if (next != null && ! next.getBlock().getValue().equals(pList[i].getBlock().getValue())) {

A similar piece of code exists on line 1066 that seems to be fixed:

                           if (next != null && next.getBlock() != null && next.getBlock().getValue() != null &&
                                    ! next.getBlock().getValue().equals(pList[i].getBlock().getValue())) {

The history seems to show that both of these expressions were changed on 29 March 2022 and the second expression was then updated on 26 July 2022. See diff in https://github.com/JMRI/JMRI/commit/9285ebca0fbb0386de2cfe438e4d1824177e2457

Please fix line 1052 in the same way as the code on line 1066.

icklesteve, 2022-10-11:
Hi @SvenRosvall , thanks for reporting.

I see what you mean in the code, have raised PR #11510

Steve.

danielb987, 2022-10-13:
PR #11510 is approved and merged so it will be included in the next test release.

PR #11510 (2022-10-13): Resolve Block Tracking NPE
Fixes #11509
closed 2022-10-13 by danielb987