JMRI/JMRI #11511: LogixNG Global Varibles "browse" function

devel-bobm, 2022-10-11:
In JMRI 5.1.5, the LogixNG function to "Browse" global variables is not very useful in its current form.

I expect that the feature is primarily useful when sharing LogixNG solutions with others.  But the information it currently provides doesn't give much utility.

Some shortcomings I see:

- It provides data for one global variable at a time, and when it does so, it displays the "system name" without providing the variable's "user name" in the "table" portion of the window. (It does show the user name in the header portion of the window though.)  It would be nice instead to be able to "browse" the complete _set_ of variables, not just one-at-a-time.

- If you copy to the clipboard, you only get the information from the "table" portion of the window, _without_ the user name.  Copying the system name to the clipboard does not seem to me to provide much utility.

- The "print system names" checkbox has no effect.  

- It might make sense to display the variable's _user name_ in the "tables" area of the display, and only show the system name if the checkbox is checked.

- If feasible, it would be useful if it could show the current "type" associated with the global variable and its "current value" (where it makes sense).

dsand47, 2022-10-11:
The current "browse" content is deficient.

Since the "browse" selection is row specific, a tool could be created to list all of the global variables.  It should be possible to use "Log Local Variables" as a model.

[image]

```
11:40:36,976 actions.LogLocalVariables             WARN  - Log local variables: [JMRI LogixNGThread]
11:40:36,976 actions.LogLocalVariables             WARN  -     Name: col, Value: 4 [JMRI LogixNGThread]
11:40:36,977 actions.LogLocalVariables             WARN  -     Name: name, Value: r3c3 [JMRI LogixNGThread]
11:40:36,977 actions.LogLocalVariables             WARN  -     Name: row, Value: 4 [JMRI LogixNGThread]
11:40:36,977 actions.LogLocalVariables             WARN  - Global variables: [JMRI LogixNGThread]
11:40:36,977 actions.LogLocalVariables             WARN  -     Global Name: g_sensor_map, [JMRI LogixNGThread]
11:40:36,978 actions.LogLocalVariables             WARN  -         r2c3 -> 2, [JMRI LogixNGThread]
11:40:36,978 actions.LogLocalVariables             WARN  -         r3c2 -> 3, [JMRI LogixNGThread]
11:40:36,978 actions.LogLocalVariables             WARN  -         r3c3 -> 3, [JMRI LogixNGThread]
11:40:36,978 actions.LogLocalVariables             WARN  -         r1c1 -> 1, [JMRI LogixNGThread]
11:40:36,978 actions.LogLocalVariables             WARN  -         r1c2 -> 1, [JMRI LogixNGThread]
11:40:36,978 actions.LogLocalVariables             WARN  -         r2c1 -> 2, [JMRI LogixNGThread]
11:40:36,979 actions.LogLocalVariables             WARN  -         r1c3 -> 1, [JMRI LogixNGThread]
11:40:36,979 actions.LogLocalVariables             WARN  -         r2c2 -> 2, [JMRI LogixNGThread]
11:40:36,979 actions.LogLocalVariables             WARN  -         r3c1 -> 3, [JMRI LogixNGThread]
11:40:36,979 actions.LogLocalVariables             WARN  -     Global Name: g_sensor_list, [JMRI LogixNGThread]
11:40:36,979 actions.LogLocalVariables             WARN  -         0: r1c1, [JMRI LogixNGThread]
11:40:36,980 actions.LogLocalVariables             WARN  -         1: r1c2, [JMRI LogixNGThread]
11:40:36,981 actions.LogLocalVariables             WARN  -         2: r1c3, [JMRI LogixNGThread]
11:40:36,981 actions.LogLocalVariables             WARN  -         3: r2c1, [JMRI LogixNGThread]
11:40:36,981 actions.LogLocalVariables             WARN  -         4: r2c2, [JMRI LogixNGThread]
11:40:36,982 actions.LogLocalVariables             WARN  -         5: r2c3, [JMRI LogixNGThread]
11:40:36,982 actions.LogLocalVariables             WARN  -         6: r3c1, [JMRI LogixNGThread]
11:40:36,983 actions.LogLocalVariables             WARN  -         7: r3c2, [JMRI LogixNGThread]
11:40:36,983 actions.LogLocalVariables             WARN  -         8: r3c3, [JMRI LogixNGThread]
11:40:36,983 actions.LogLocalVariables             WARN  - Log local variables done [JMRI LogixNGThread]
```

dsand47, 2022-10-11:
I think this covers the issues for the row specific browse action.
[image]

devel-bobm, 2022-10-11:
I like the concept!  But it could use some refinement.

- JMRI 5.1.5's code, and your modified version, gives the window a name of "LogixNG/Module Browser", even though the window is browsing LogixNG "global variable"!

- The data displayed needs to formatted in a way suitable to the variable type.  The displayed content shown in the image above would be appropriate if the variable were of type "Array", but it instead is of type "Map".  For a Map, it would be more appropriate to display "key->value" pairs.

- It might be useful to be able to _selectively_ prune the data.  A checkbox at the bottom to limit the length of a displayed value to (n) characters, with (n) defined by the developer, and applied to each data element in an Array, or each "Key" or each "value" in a Map, or ...  
 
- Reporting all of that data "exactly" would result in a difficult-to-read display for those cases where Maps/Arrays were large and where variable content/keys/values were "long".  In the particular LogixNG application I was working with, an exact representation of the data would have been "ugly", simply because the Map I worked with included String "Key"s which were about 35 characters long - about 60 of them!  Being able to "shorten" the information would be useful in some cases, while an "exact" representation would be useful in other cases.  Perhaps another "checkbox" at the bottom of the window to enable a "shorten long strings" (to some arbitrarily-chosen length) option?

dsand47, 2022-10-11:
The initial release of LogixNG only had the browser feature for LogixNG and Module.  The window title can be fixed.  The options at the bottom of the window should be relevant to the LogixNG table type.

Expanding the Array and Map data is already done.

[image]

Trimming the content to a fixed or optional width could be done, but does it apply to all browsers, including the LogixNG and Module trees?

devel-bobm, 2022-10-11:
Oh, my misunderstanding! I did not notice that the _keys_ you used in your Map were of the form "rXcY", and not implying "rows"/"columns" within the Map!  I should have looked more-carefully at the text from the "log local variables" shown previously!

Yes, this meets my "minimal requirements" for the Global Variable browse feature. It is a HUGE improvement over previous behavior for Global Variables of "Map" type!