JMRI/JMRI #11520: JMRI not remembering throttle state?

Aikagi000, 2022-10-15:
Hello All,

I'm currently running JMRI v5.0 with java 11.0.12. and connected to DCC++
in short. If I select a loco, press a function, release the loco, then  reacquire the loco, JMRI doesn't remember that the function was already activated. this goes for the speed setting as well.

so I can set a loco running, drop it to do switching with a diff loco, then pick up the first loco and it would say speed zero with no functions on. I would then have to remember what buttons are pushed and reset the speed. Did I miss a setting? or is this a bug? Cause I know it used to remember, which was why I stopped using my power cab in the first place.

Thanks and best,

Lionel642, 2022-10-17:
Hello;

Just tested this with my Digitrax system and it works fine (throttle state
is initialized from command station state), so I guess it's the DCC+
throttle implementation in JMRI that does not read throttle state when
acquiring it (if it does support it).

Regards;
Lionel

danielb987, 2022-10-17:
@mstevetodd 
Is this something for you to look at?

Aikagi000, 2022-10-18:
@Lionel642 
Ah ok, I didn't know it had to do with the command station. Now it makes sense why it's across everything (Jmri throttles and also my train driver throttles). 

On thing that is also a part of the issue, if I have an open jmri throttle for train A on jmri and my phone, the functions will mirror each other. If I drop train A from my phone and do switching, when i pick train A back up it will pick up the old settings only because the jmri throttle was left open.

closed 2022-12-18 by stale[bot]

Aikagi000, 2023-01-14:
Sight update. Each loco would have to be temp added to it's own consist and consist 0 is already in use it seems. Also I had a weird bug in that when I quit JMRI (without turning off track power) the track power was still on and locos were still running. It seems using the consist tool is writing to memory somewhere in the DCC++ base station. It wasn't until I restarted JMRI that power cut off. It really seems like it's remembering the state of all the functions after using the consisting tool. There is something in this and that consist tool has the key written it in.