JMRI/JMRI #11542: Unable to view web content css strong class="button

icklesteve, 2022-10-27:
See missing text highlighted in red.

html content eg. 
`<p>Click <strong class="button">Available Plugins</strong></p>`

I'm not sure how many other pages this affects, however in default.css, white may not be a good default for button text when mainContent background is also white.

Likely easier to change the class applied to the `<strong>` as it's not actually a button, it may be better described elsewhere.

[image]

silverailscolo, 2022-10-28:
Mind a <div class="button"> is part of the JMRI website navigation, on a filled text. A fix for this issue should leave that in place.

danielb987, 2022-10-29:
@silverailscolo 
I edited your reply to make `<div class="button">` visible in your comment.

bobjacobsen, 2023-03-29:
@icklesteve The [NetBeans doc page](https://www.jmri.org/help/en/html/doc/Technical/NetBeans.shtml) show above seems to have this problem resolved.  Can we close the issue?

closed 2023-05-31 by icklesteve