JMRI/JMRI #11567: Adoptium JDK 11 isn't detected by installer

SirTainly, 2022-11-05:
If you install Adoptium JDK from the zip file, you don't get the registry keys set on Windows, which prevents JMRI from installing as it can't find Java 11. It might be better to relax this check, or check that a java executable does exist rather than relying on the registry.

crosland, 2022-11-05:
I seem to recall something similar (I think with OpenJDK) but the installer for that had an option to set the correct registry keys.

bobjacobsen, 2022-11-05:
JMRI’s java install page says "On Windows: when you run the Java installer, be sure to select the "JavaSoft (Oracle) registry keys" option.”  Is that not correct when running from the .zip file?

https://www.jmri.org/java/

devel-bobm, 2022-11-05:
@crosland, I believe it was the Azul OpenJDK 11 that had the problem, and it was solvable by using the .MSI installation package and clicking the appropriate option during setup.  There was _no_ solution for using a .zip install of Azul, other than perhaps some (risky) Registry Editor work.

@SirTainly,
The next "test release" of JMRI, version 5.1.6, should have a way to "work around" the problem.  See [https://www.jmri.org/install/WindowsNew.shtml#installation](url) .  I would expect this release to happen in the next few days, barring unforseen complications.

@bobjacobsen, you are (typically) correct.  Simply unzipping a JAVA package will not, in and of itself, modify any of the Windows Registry entries which have been required for verifying the JAVA install during the JMRI Window installation process, or by the JMRI Windows "Launcher".

SirTainly, 2022-11-05:
@devel-bobm thanks, will check it out once 5.1.6 is available

To confirm, I managed to get around it by installing a different java 11 from an .msi probably Azul.

@bobjacobsen There is no installer, you just unzip the jdk, set JAVA_HOME and PATH and off you go, so nothing is setting registry keys. Tbh this has been my preferred way of dealing with java installs since I don't like things modifying PATHs/registry etc for me. Finding out something else has switched which jdk is first on the path has caused me no end of hair loss in 20+ years of professional java development! :)

devel-bobm, 2022-11-05:
@SirTainly,

JMRI 5.1.6 has been released, per [https://groups.io/g/jmriusers/message/210004](url) .

Please update this issue regarding whether you were able to make use of those new JMRI features related to JMRI installation and launch on Windows platform machines.

SirTainly, 2022-11-11:
Apologies for the delay, I'm away on business, should be home and able to
retest this weekend

closed 2023-01-11 by stale[bot]