JMRI/JMRI #11604: ActionCreateBeansFromTable.java issue

dsand47, 2022-11-23:
The `Update beans to user name` option does not work when doing the Move option.  The NamedBeanHandle references are not updated.

The update option should rarely be needed.  It is only relevant when a user name has been assigned to a bean and the `No` option was selected on the dialog to change the reference to the user name.

The problem appears to be in the moveUserName method.  The current approach does the `moveBean` followed by the `updateBeanFromSystemToUser`.   The problem is that those methods along with `inUse` do nothing since the reference is based on the system name, not the user name.

The following change does the `updateBeanFromSystemToUser` first after checking the `old` values instead of the `new` values.

Now the bean references are based on the user name and will behave like any other bean reference.

I don't know if this is the best solution but it does work, at least using my simple test case.

```
diff --git a/java/src/jmri/jmrit/logixng/actions/ActionCreateBeansFromTable.java b/java/src/jmri/jmrit/logixng/actions/ActionCreateBeansFromTable.java
index b6d8d3073f..5573d9db9b 100644
--- a/java/src/jmri/jmrit/logixng/actions/ActionCreateBeansFromTable.java
+++ b/java/src/jmri/jmrit/logixng/actions/ActionCreateBeansFromTable.java
@@ -295,14 +295,21 @@ public class ActionCreateBeansFromTable extends AbstractDigitalAction
             throws JmriException {
 
         NamedBeanHandleManager nbMan = InstanceManager.getDefault(NamedBeanHandleManager.class);
-        oldNameBean.setUserName(null);
-        newNameBean.setUserName(userName);
-        nbMan.moveBean(oldNameBean, newNameBean, userName);
-        if (nbMan.inUse(newNameBean.getSystemName(), newNameBean)) {
+
+        if (nbMan.inUse(oldNameBean.getSystemName(), oldNameBean)) {
             if (_updateToUserName) {
-                nbMan.updateBeanFromSystemToUser(newNameBean);
+                nbMan.updateBeanFromSystemToUser(oldNameBean);
             }
         }
+
+        oldNameBean.setUserName(null);
+        newNameBean.setUserName(userName);
+        nbMan.moveBean(oldNameBean, newNameBean, userName);
+//         if (nbMan.inUse(newNameBean.getSystemName(), newNameBean)) {
+//             if (_updateToUserName) {
+//                 nbMan.updateBeanFromSystemToUser(newNameBean);
+//             }
+//         }
     }
 
     /** {@inheritDoc} */
```

dsand47, 2022-11-23:
There is a bug in ActionFindTableRowOrColumn.java.
```
diff --git a/java/src/jmri/jmrit/logixng/actions/ActionFindTableRowOrColumn.java b/java/src/jmri/jmrit/logixng/actions/ActionFindTableRowOrColumn.java
index 13c89e7d77..e30468c00e 100644
--- a/java/src/jmri/jmrit/logixng/actions/ActionFindTableRowOrColumn.java
+++ b/java/src/jmri/jmrit/logixng/actions/ActionFindTableRowOrColumn.java
@@ -188,7 +188,7 @@ public class ActionFindTableRowOrColumn extends AbstractDigitalAction
                     if ((cell != null) && (cell.equals(value))) {
                         Map<String, Object> columnData = new HashMap<>();
 
-                        for (int colIndex=1; colIndex <= table.numRows(); colIndex++) {
+                        for (int colIndex=1; colIndex <= table.numColumns(); colIndex++) {
                             Object subHeader = table.getCell(0, colIndex);
                             if ((subHeader != null) && (!subHeader.toString().isEmpty())) {
                                 columnData.put(subHeader.toString(), table.getCell(row, colIndex));
```

danielb987, 2022-11-23:
Thanks. I'm currently offline and only using a mobile phone so feel free to create a PR for these two issues if you want to.

I can review the PR with the GitHub app on the phone.

closed 2022-11-24 by danielb987