JMRI/JMRI #11608: Zimo sound decoder files update

nigelcliffe, 2022-11-24:
I've updated various Zimo files, in the attachments.   Can someone please take this and move it into a test release.

1 - update to MX-series to add MX660 decoder,  replace the decoder file "  Zimo_Unified_software_v40_MX64x_MX65x.xml " with new file. 

2 - Reworked the Zimo MS-decoder file to have a more sensible sub-component structure.   Replace the decoder file  " Zimo_MS_small_v4.xml"   with new version  (requires step 3 for this file to work) 

3)   place contents of "zimo folder files" into the "zimo" sub-folder in the decoder files.   This contains some replacement files and some new files.   Contains an update to three decoder panes to add additional features, and revised structure of CV components for the MS-series decoders.  

4)  please DELETE the following files from the "zimo" sub-folder as no longer required:     ( these files were created in a previous 5.1x test release, removing them to improve clarity and prevent subsequent people trying to edit obsolete files)  
       CVs_MS_series.xml
       CV400-CV428_MS_series.xml
       CV100-CV152_MS_series,xml
       CV260-CV355sound_MS_series.xml
       CV372-CV399sound_MS_series.xml
       CVSwissMapping_MS_series.xml

[Zimo_decoder_files.zip](https://github.com/JMRI/JMRI/files/10085063/Zimo_decoder_files.zip)

[Zimo_folder_files.zip](https://github.com/JMRI/JMRI/files/10085061/Zimo_folder_files.zip)

bobjacobsen, 2022-11-25:
Happy to do this.  But I'm not seeing these two files to delete:

CVs_MS_series.xml
CV400-CV428_MS_series.xml

Are those correct?

nigelcliffe, 2022-11-25:
Sorry Bob, 

my mistake.    the last two you mention ( CVs_MS_series.xml ,  CV400-CV428_MS_series.xml  ) were in my local files, only.   They are things I created before deciding to tidy everything, but hadn't submitted to the JMRI project.   

Ignore those two from the request. 

- Nigel

closed 2022-11-26 by danielb987