JMRI/JMRI #11612: jmri.jmrit.dispatcher.LoadAtStartUpTest has lots of CI failures

danielb987, 2022-11-26:
Most of CI succeeds today which is very good. But `jmri.jmrit.dispatcher.LoadAtStartUpTest` has a lot of failures.

https://github.com/JMRI/JMRI/actions/runs/3544409023/jobs/5951654823
```
2022-11-25T00:38:15.4059380Z   JUnit Jupiter:LoadAtStartUpTest:testShowAndClose()
2022-11-25T00:38:15.4060400Z     MethodSource [className = 'jmri.jmrit.dispatcher.LoadAtStartUpTest', methodName = 'testShowAndClose', methodParameterTypes = '']
2022-11-25T00:38:15.4069230Z     => org.opentest4j.AssertionFailedError: "Continuing speed was 0.1 not 0.35" did not occur in time
2022-11-25T00:38:15.4069930Z        org.junit.jupiter.api.AssertionUtils.fail(AssertionUtils.java:38)
2022-11-25T00:38:15.4071870Z        org.junit.jupiter.api.Assertions.fail(Assertions.java:135)
2022-11-25T00:38:15.4073490Z        jmri.util.JUnitUtil.waitFor(JUnitUtil.java:490)
2022-11-25T00:38:15.4074170Z        jmri.jmrit.dispatcher.LoadAtStartUpTest.testShowAndClose(LoadAtStartUpTest.java:217)
```

https://github.com/JMRI/JMRI/actions/runs/3544409023/jobs/5955965583
```
2022-11-25T09:19:11.7677890Z   JUnit Jupiter:LoadAtStartUpTest:testShowAndClose()
2022-11-25T09:19:11.7679100Z     MethodSource [className = 'jmri.jmrit.dispatcher.LoadAtStartUpTest', methodName = 'testShowAndClose', methodParameterTypes = '']
2022-11-25T09:19:11.7682570Z     => org.opentest4j.AssertionFailedError: "Continuing speed was 0.1 not 0.35" did not occur in time
2022-11-25T09:19:11.7692960Z        org.junit.jupiter.api.AssertionUtils.fail(AssertionUtils.java:38)
2022-11-25T09:19:11.7707680Z        org.junit.jupiter.api.Assertions.fail(Assertions.java:135)
2022-11-25T09:19:11.7708690Z        jmri.util.JUnitUtil.waitFor(JUnitUtil.java:490)
2022-11-25T09:19:11.7709290Z        jmri.jmrit.dispatcher.LoadAtStartUpTest.testShowAndClose(LoadAtStartUpTest.java:217)
```

https://github.com/JMRI/JMRI/actions/runs/3544409023/jobs/5963670423
```
2022-11-25T19:50:46.9525360Z   JUnit Jupiter:LoadAtStartUpTest:testShowAndClose()
2022-11-25T19:50:46.9526400Z     MethodSource [className = 'jmri.jmrit.dispatcher.LoadAtStartUpTest', methodName = 'testShowAndClose', methodParameterTypes = '']
2022-11-25T19:50:46.9526950Z     => org.opentest4j.AssertionFailedError: "Continuing speed was 0.1 not 0.35" did not occur in time
2022-11-25T19:50:46.9527380Z        org.junit.jupiter.api.AssertionUtils.fail(AssertionUtils.java:38)
2022-11-25T19:50:46.9527770Z        org.junit.jupiter.api.Assertions.fail(Assertions.java:135)
2022-11-25T19:50:46.9528120Z        jmri.util.JUnitUtil.waitFor(JUnitUtil.java:490)
2022-11-25T19:50:46.9528550Z        jmri.jmrit.dispatcher.LoadAtStartUpTest.testShowAndClose(LoadAtStartUpTest.java:217)
```

https://github.com/JMRI/JMRI/actions/runs/3544409023/jobs/5968436221
```
2022-11-26T09:31:12.6949080Z   JUnit Jupiter:LoadAtStartUpTest:testShowAndClose()
2022-11-26T09:31:12.6967920Z     MethodSource [className = 'jmri.jmrit.dispatcher.LoadAtStartUpTest', methodName = 'testShowAndClose', methodParameterTypes = '']
2022-11-26T09:31:12.6968900Z     => org.opentest4j.AssertionFailedError: "Speed is correct" did not occur in time
2022-11-26T09:31:12.6969670Z        org.junit.jupiter.api.AssertionUtils.fail(AssertionUtils.java:38)
2022-11-26T09:31:12.6970370Z        org.junit.jupiter.api.Assertions.fail(Assertions.java:135)
2022-11-26T09:31:12.6970950Z        jmri.util.JUnitUtil.waitFor(JUnitUtil.java:490)
2022-11-26T09:31:12.6971680Z        jmri.jmrit.dispatcher.LoadAtStartUpTest.testShowAndClose(LoadAtStartUpTest.java:182)
```

@Pugwash1 
Do you have any idea on what we can do about this?

Pugwash1, 2022-11-26:
The line 182 errors every time on my PC so I can look at that. But won't be till late today.
Failing on a wait for that we have just asserted is correct is a little wild

danielb987, 2022-11-26:
> The line 182 errors every time on my PC so I can look at that. But won't be till late today. Failing on a wait for that we have just asserted is correct is a little wild

It's not urgent. But it would be good to fix.

See PR #11563. I added the waitFor() since the test was failing later. Maybe  #11563 should be reverted?

Pugwash1, 2022-11-26:
@danielb987 see if  #11613 works for you

danielb987, 2022-11-28:
> @danielb987 see if #11613 works for you

It seems to work very well! Thanks!

danielb987, 2022-11-29:
Closing this since it seems to work much better now.
Thanks @Pugwash1 !

closed 2022-11-29 by danielb987