JMRI/JMRI #11637: Start up Wizard runs whenever I start decoder Pro

Min31, 2022-12-06:
I have installed JMRI and added quite a few locos to the roster.  However whenever I start Decoder Pro it takes me through the Start Up Wizard again.  A second problem I have is that the roster list row height is so small I can't read the individual lines.

dsand47, 2022-12-06:
Go to Preferences => Display and make the font size is not set to zero.

Dave Sand

Min31, 2022-12-06:
Perhaps it's related to the fact that the Start Up Wizard runs every time I start decoder pro?

Min31, 2022-12-06:
I have tried again.  Your suggestion is correct the font size is in fact zero but as I say it requires that I restart Decoder Pro for it to take effect and that takes me back to the Start Up Wizard again which resets it to 0.  ??

dsand47, 2022-12-06:
You need to post the contents the "session.log" file.  When you have DecoderPro running, go to "Help => File Locations".  Click on the "Open Log Files Location".  This will open the file manager at the directory that contains the session.log file.  Now Quit DecoderPro.

Attach the session.log file in a reply.

Dave Sand

dsand47, 2022-12-06:
On second thought, attached the "messages.log" file.

Dave Sand

Min31, 2022-12-06:
Here is the file
[session.log](https://github.com/JMRI/JMRI/files/10168865/session.log)

Min31, 2022-12-06:
Messages log file attached..

Min31, 2022-12-06:
No idea Is it something to do with 64 bit 32 bit apps

dsand47, 2022-12-06:
After looking at the rest of the messages.log, it appears that you have tried a number of things that did not resolve the issues.

I recommend a clean re-start.

- Reset the JMRI install.
  - Use the Windows control panel to remove JMRI.
  - Delete the residual JMRI directory at C:\Program Files (x86).
  - Install a fresh download of JMRI 5.0.
- Reset the JMRI user data.
  - Rename C:\Users\Russell\JMRI to something like JMRIsaveDir.
  - Start DecoderPro and do the Start Up Wizard.
  - After the restart, verify the font size in Preferences => Display.
  - Stop DecoderPro
  - Attach the current messages.log file.

Min31, 2022-12-06:
Thanks will do

Min31, 2022-12-06:
Thanks will do and advise

Min31, 2022-12-06:
I have to go out for some time and will get back to you when I return

Min31, 2022-12-06:
Log file attached

[messages.log](https://github.com/JMRI/JMRI/files/10170192/messages.log)

Min31, 2022-12-07:
This is after the JMRI install.

I had completely uninstalled JMRI and renamed the file.  Don't recall if I installed for all users.

I did reboot the computer

Min31, 2022-12-07:
sorry pressed the wrong button when closing

Min31, 2022-12-07:
I sent it to your email. Can't work out to attach to these messages

dsand47, 2022-12-07:
The last file I received was sent via GitHub 7 hours ago, around 7:30 AM your time.  It was the same file you sent earlier which means that had not done the directory rename.

An alternate approach to attaching the messages.log file is to open it with WordPad and copy and paste the contents.

dsand47, 2022-12-07:
The messages.log file looks exactly as expected.  You now have a clean user data configuration.

Where you able to set the font size using Preferences => Display and have the change work?

Min31, 2022-12-07:
Yes that change of font worked

Min31, 2022-12-07:
I will now try to load some locos and set them up.

Thank you for your help and patience with me.

Min31, 2022-12-07:
Dave I have loaded and programmed many locos.  All working perfectly.
Thanks again for your help.

Russell

icklesteve, 2022-12-08:
see #11641 which sets font size to default in the new user profile created by the DecoderPro Wizard.

closed 2023-02-10 by stale[bot]