JMRI/JMRI #11653: Feature: Support for CanDigitalBahn CC-Schnitte 2.1 Command Station

ralflang, 2022-12-16:
# CC-Schnitte 2.1 support for Marklin systems

The CC-Schnitte is a third party USB interface for the Marklin CAN Bus. It is usually directly attached to the 60113 Digital Connector Box, with the MobileStation 2 attached to the second port. In the Marklin system, the Connector Box receives commands from devices attached to the CAN Bus. It acts as a mini booster, but also translates them into a proper DCC/mm2/mfx signal. CC-Schnitte combines with the Connector Box in a fashion similar as the already supported CentralStation 2

## Implementation idea

I have had a look at both the JMRI Marklin CS2 implementation in current master branch and two other opensource implementations for CC-Schnitte support in other products. From what I understand the CAN signals in CC-Schnitte 2.1 and in CS2 are fundamentally the same, but wrapped in USB rather than TCP/IP over ethernet.

I think I can mimic other stations' serial support code and wrap it around the marklin code to easily support the CC-Schnitte, reusing existing JMRI code for Marklin CAN.

Before I head off with a fork and PR I am opening this issue to allow for discussion. I am only recently reading JMRI code and I want to benefit from others' experience where possible.

## Driver name

I suggest vendor "CdB" and product "ccschnitte" should be appropriate

## Contributor Agreement

I understand I must sign some CA or other legalese in order to contribute. I will read up on this once I see this works at all.

danielb987, 2022-12-17:
@ralflang 
Welcome to the JMRI project!

> Before I head off with a fork and PR I am opening this issue to allow for discussion. I am only recently reading JMRI code and I want to benefit from others' experience where possible.

It might be good to write a message to the [JMRI developers list](https://jmri-developers.groups.io/g/jmri/topics).  [Join here](https://jmri-developers.groups.io/g/jmri/join).

> I understand I must sign some CA or other legalese in order to contribute. I will read up on this once I see this works at all.

@bobjacobsen 
Can you answer this?

bobjacobsen, 2022-12-17:
> I understand I must sign some CA or other legalese in order to contribute.

So long as you've accepted the GitHub terms of service and submit your changes via a PR, the paperwork is already all set

JMRI's licensing terms are laid out here:  https://www.jmri.org/copyright.shtml

You'll be providing your code to us under the GNU General Public License (GPL) version 2.0, and we'll distribute it further under that same license.

ralflang, 2022-12-17:
Yes, that's all well. I just saw that section on your main site and somehow got it wrong. I am used to contributing to GPLed projects. Tried to register to the ML and sent a mail but it seems like it needs moderation or the account needs a first time admin approval.

I'm looking forward to actually building that feature. :)

bobjacobsen, 2023-03-22:
PR #11852 with initial support has been merged, and is in JMRI 5.3.5