JMRI/JMRI #11655: Font is Tiny

scottso699, 2022-12-17:
[image]
[profile.zip](https://github.com/JMRI/JMRI/files/10250393/profile.zip)
[session.log](https://github.com/JMRI/JMRI/files/10250394/session.log)
[systemInfo.txt](https://github.com/JMRI/JMRI/files/10250395/systemInfo.txt)

Issue Type: __Bug__

I am seeing an issue with the font in the JMRI Decoder Pro ever since I upgraded to Windows11 (Not intentionally). Now the fonts in the programming GUIs are tiny and odd looking. Compatibility issue between Java 11 and Windows 11? Any ideas on how to correct?

Thanks,

Scott Upmal
scott.upmal@gmail.com

DecoderPro 5.1.8+R71fbf5dc7e
OpenJDK 64-Bit Server VM OpenLogic 11.0.17+8-adhoc..jdk11u mixed mode
Windows 11 10.0 amd64

Pugwash1, 2022-12-17:
@scottso699  use preferences - Display. See if one of the choices windows, windows classic, gtk works better, also check the font you are using, if it's not available then an ugly font may have been used instead.

dsand47, 2022-12-17:
Scott,

Use Preferences => Display and change the font size if zero. 

Dave Sand

JMRI Screenshot 2022-12-16 200530 <https://user-images.githubusercontent.com/120762187/208215593-db266fb2-cf2b-4fe2-8113-ad3227584a5b.jpg>

closed 2023-02-17 by stale[bot]