JMRI/JMRI #11657: Error 404

dennismanson2116, 2022-12-17:
https://www.jmri.org/community/clinics/NMRA2008/index.html

Thanks for the great software!
dennismanson2116@gmail.com

PUlvestad, 2022-12-18:
Where did you find that url?

The correct url is https://www.jmri.org/community/clinics/NMRA2008/index.shtml

danielb987, 2022-12-18:
@dennismanson2116 
Welcome to the JMRI project!

Your last reply didn't had any extra information. Do you know where you found this link?

dennismanson2116, 2022-12-19:
I thought the screen shot would help. It was the page that discussed how to
learn the panel editor and offered a link to a pdf file that someone had
put together as a tutorial.
Sorry, the url just showed www.jmri.org

V/R
Dennis

danielb987, 2022-12-19:
> I thought the screen shot would help. It was the page that discussed how to learn the panel editor and offered a link to a pdf file that someone had put together as a tutorial. Sorry, the url just showed [www.jmri.org](http://www.jmri.org) V/R Dennis

It's not possible to attach a file to an email to post it at GitHub. The attached file is removed from the message.

If you want to attach an image to a message on a GitHub Issue, you need to use the web browser instead.

I'm still not able to find the page with the link to https://www.jmri.org/community/clinics/NMRA2008/index.html

dennismanson2116, 2022-12-19:
Oh, well. Maybe someone else can better help. I just thought you might like
to know.

V/R
Dennis

danielb987, 2022-12-19:
> Oh, well. Maybe someone else can better help. I just thought you might like to know.

Yes, we want to know! If we just know which page has the bad link, it's easy to fix it. But please tell us which page has the bad link.

PUlvestad, 2022-12-19:
Daniel,
Not able to find it yet but
https://www.jmri.org/help/en/html/apps/PanelPro/index.shtml#pe
has 3 invalid url's

dsand47, 2022-12-19:
It is in help/en/package/jmri/jmrit/display/LayoutEditor.shtml, line 169.

PR #11660 (2022-12-20): update NMRA2008 clinic weblinks
fixes #11657 
https://jmri.org/community/clinics/NMRA2008/index.html to https://jmri.org/community/clinics/NMRA2008/index.shtml
closed 2022-12-20 by danielb987