JMRI/JMRI #11658: [Feature request] Providing a Flatpak version of JMRI for GNU/Linux systems

CyberFox001, 2022-12-18:
Hello,

This is just a suggestion. To distributing JMRI for GNU/Linux systems,
it would be nice to use Flatpak.

Flatpak is a package format and a system to build and run desktop
application on GNU/Linux systems. It provide a way to defining a
precise and reproducible environment for the application, is a free
(as in freedom) software and it's available on a lot of GNU/Linux like
Ubuntu, Debian, Fedora, Linux mint, CentOS, etc.

By providing a Flatpak, it will simplify the installation and update
of JMRI. Especially regarding the Java version needed.

All the developer documentation about Flatpak can be found here:
https://docs.flatpak.org/en/latest/

For distributing the Flatpak package, you can use Flathub.org. It's a
community repository that distribute a lot of application in Flatpak
format.

The Flathub website is: https://flathub.org/
And the publication guide is on their wiki: https://github.com/flathub/flathub/wiki/App-Submission

After released on Flathub, users can install, update and remove JMRI from their software manager.

arcDaniel84, 2022-12-28:
+1 absolute great Idea

arcDaniel84, 2023-01-05:
There are other application on flathub using java and for JMRI the openJDK is sufficient

CyberFox001, 2023-02-20:
@arcDaniel84: Do you have a list some of the FOSS applications on Flathub using java ?

We can take inspiration from it.

bobjacobsen, 2023-03-29:
I've tried to get `flatpak-builder` to work on my macOS system without success. I think we need somebody who's linux-based in order to see how hard this will be.

CyberFox001, 2023-05-18:
Any new ?

bobjacobsen, 2023-05-18:
Nobody's been identified as both (1) wanting to work on it and (2) having the linux machine and skills to be able to.

You might want to ask about the interest on the jmriusers group.

dsand47, 2023-05-18:
I did a brief review of the process.  It is a lot of work to eliminate a web download and 4 terminal commands.

CyberFox001, 2025-01-23:
I took the time to write a Flatpak manifest to build a Flatpak version of JMRI.

It's still a work in progress:
https://framagit.org/SebGen/jmri-flatpak

I'm currently working on the icons and AppStream metada.
It launch, have access to user configuration but I haven't tested yet on my own model railway.

Do JMRI have a PNG or SVG version of its application icons ?

bobjacobsen, 2025-01-23:
> On Jan 23, 2025, at 11:53 AM, CyberFox001 ***@***.***> wrote:
> 
> Do JMRI have a PNG or SVG version of its application icons ?

There’s .png for the LccPro icons, but all the others are .gifs.

https://www.jmri.org/resources/

and the resources/ directory

danielb987, 2025-01-23:
> [Spårningsbild][https://github.com/notifications/beacon/AE2REVJ3URTEIIH2KFQL6XT2MFD3BA5CNFSM6AAAAABVXZ4BL6WGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTU3T5XKS.gif]

Creating a png from a gif should be easy. Only store the gif as a png in a drawing program, like Gimp.

Daniel

bobjacobsen, 2025-01-24:
> About the Java version: I have used OpenJDK 11. But if you prefer, I can use the version 17.

Java 17 would be better.  Which JMRI release is this based on?

> About the distribution to user:
> Do you prefer to review what I have done first ?

I'm not in a position to review this, sorry.  Perhaps some linux user can try it?

> Do we publish as a "Flatpak beta" and wait for feedback ?
> How do you prefer to distribute: Flathub repository, a private repo or a single bundle file to download ?
> Note: If published on Flathub, it would be more discoverable by new users on Linux through their application managers.

How often do you want to update this? 

We can add the single file to a JMRI release page (e.g. as another asset on https://github.com/JMRI/JMRI/releases/tag/v5.11.2) 

But Flathub also looks fine.

CyberFox001, 2025-01-25:
> Java 17 would be better. Which JMRI release is this based on?

Ok, I have switched Java from version 11 to 17.

For JMRI, it's based on version 5.11.2.

> How often do you want to update this?

I hope to update on each release of JRMR. If the update is not major, it need a few minutes to be done.
But, as a student, I cannot guarantee to have the needed amount of time the day of a release. So it could have a delay.
And I need to find a way to be notified when a new release is done.

bobjacobsen, 2025-01-25:
>>How often do you want to update this?

> I hope to update on each release of JRMR. If the update is not major, it need a few minutes to be done.
> But, as a student, I cannot guarantee to have the needed amount of time the day of a release. So it could have a delay.

Can a shell script be written that takes some pre-created files, some arguments and a JMRI install .tgz and produces a new Flatpack file?  E.g. takes an existing Flatpak file and updates the JMRI file within it. 

If so, we can run that as part of our semi-automated build system for creating new releases.

> And I need to find a way to be notified when a new release is done.

If you to subscribe to notifications from the JMRI/JMRI repository, you'll get a notification when a release is created.  Two problems with that:
 - You'll get lots of other notifications too.  There doesn't seem to be a way to get notifications _just_ about releases.  Though maybe your email client can filter those out.
 - You'll only hear _after_ the release is final.  That's a bit late.  Better would be to automate the process so we can release all the files at the same time.

CyberFox001, 2025-01-25:
> Can a shell script be written that takes some pre-created files, some arguments and a JMRI install .tgz and produces a new Flatpack file? E.g. takes an existing Flatpak file and updates the JMRI [...]

The actual Flatpak version will take the source code from the Git repo and build it with an `ant` command. The version is selected by the Git tag written inside the Flatpak manifest.

To do an update, it require at minimum to modify this tag in the manifest file. I can check if their is a way to capture and use this tag automatically.

An optional thing to do when an update occur, is to modify the metadata to list the new features and changes.
But maybe it can be automated by a script.

Do you prefer to keep the files needed for the Flatpak version in a separate repository, or to include it in the JMRI repo as a sub-dir ?

danielb987, 2025-01-25:
Does the building of the Flatpack requires Linux? Or can it be done on Windows and Mac?

Which are the system requirements to build the Flatpack?

CyberFox001, 2025-01-25:
> Does the building of the Flatpack requires Linux? Or can it be done on Windows and Mac?
> 
> Which are the system requirements to build the Flatpack?

It require a Linux kernel, `flatpak` and `flatpak-builder` commands installed and a configured Flatpak repo that can provide:
- ` org.freedesktop.Sdk`
- `org.freedesktop.Sdk.Extension.openjdk17`

For the repo, Flathub is enough.

If you don't have Linux, build could be done with a CI configured on the Git repo.
I haven't check on Flathub side, but I think that the app you publish on Flathub are built by them.

bobjacobsen, 2025-01-26:
It's great that you're working on this.

As a first step, I recommend:
 - You create the file and make it available somewhere so we can attach it to the [5.11.2 release page](https://github.com/JMRI/JMRI/releases/tag/v5.11.2)
 - If you'd like to, put it on Flathub
 - Once it's available, we send a note to JMRIusers and JMRI-developers asking for people to try it and give feedback
 - From that, we can determine if we need additional user documentation, etc.

Assuming that's a success, to make this part of our normal infrastructure for routine release, I recommend:
 - We create a new directory in scripts/ called scripts/Flatpack
 - Inside that, we put all the reasonably-sized components that are needed to create a Flatpack version of JMRI.  Not the linux kernel or the tool set, but the Flatpack manifest file, build.xml for Ant, any necessary scripts, README.md with instructions, etc.
 - We can then configure the builds on builds.jmri.org (a Linux machine) to create the Flatpack file(s) along side the existing .dmg, .tgz and .exe files we make available for downloading and upload to GitHub as part of the semi-automated [release process](http:jmri.org/scripts/HOWTO-distribution.md).

I'm still a little unclear on what it takes to release things to Flathub.  It looks like maybe that can be part of the semi-automated JMRI release process too, but I'm not sure.

ralflang, 2025-01-29:
I could probably help building the regular flatpak updates in CI containers if needed.

CyberFox001, 2025-02-03:
I'm back. Sorry for the week with no news, it's a bit crazy here.

So, I am writing the Metadata infos. These infos will be displayed on software manager, on GNU/Linux desktops, and also on the Flathub website. With some screenshots.

I have a few questions about this part:

- Do you have a short and long description you want to use for JMRI ?
- Do you have screenshots you want to use for presenting JMRI ?
- Which licence do I apply on these metadata and the screenshots ?

> As a first step, I recommend:
>
>     * You create the file and make it available somewhere so we can attach it to the [5.11.2 release page](https://github.com/JMRI/JMRI/releases/tag/v5.11.2)
>
>     * If [...]

I can put the Flatpak app file here ? 

Do we ask people for try and give feedback before publishing on Flathub, so when we release on Flathub it have less problems ?

bobjacobsen, 2025-02-03:
> Do you have a short and long description you want to use for JMRI ?

Short: JMRI is a set of community-developed tools for configuring and running your model railroad.

Long: JMRI is a set of community-developed tools for configuring and running your model railroad. With it, you can configure DCC decoders, control the turnouts and lights on your railroad, automate their operation, build control panels, create switch lists, and if you'd like, run automated trains.  It works with most common model railroad electronics. JMRI provides both simple and very powerful tools for model railroad.

> Do you have screenshots you want to use for presenting JMRI ?

How about:
https://www.jmri.org/help/en/html/apps/DecoderPro/images/DP5Roster.png
https://www.jmri.org/community/examples/images/JMRI_QV_CTC.gif

> Which licence do I apply on these metadata and the screenshots ?

Can you refer people to our licensing page?  https://www.jmri.org/copyright.shtml  If not, then they should be marked with CC BY-SA 4.0.

> I can put the Flatpak app file here ?

Not sure what you're asking, sorry.  If you're asking about a generally-available tool for making Flatpaks, it would be better to put the instructions for obtaining it in a README file within the new scripts/Flatpack directory.  If you're asking about the Flatpak itself and whether to attach it to this Issue, it depends on how big it is. GitHub has a size limit.  You could try it and see if it'll attach.

> Do we ask people for try and give feedback before publishing on Flathub, so when we release on Flathub it have less problems ?

I think that would be wise.  We can roll it out through jmri-developers and then, if OK, jmriusers to see what the reaction is.

CyberFox001, 2025-02-23:
The bundle file is mostly done.

I have some problem to make the license and screenshots visible in the app center before installation.
But after the installation, all information are visible in the app center.

danielb987, 2025-03-01:
> Good news: I have a Flatpak file ready to test.
>
> Bad news: Gitlab do not let me easily attach an arbitrary file to a release.
>
> I need to find a way to share my file easily.

Add .txt extension to the file name and attach it to the issue.

Daniel

CyberFox001, 2025-03-02:
I tried, but Github say it failed. I think it's because of the size.

bobjacobsen, 2025-03-02:
Got it.  Could you please confirm that it's still based on JMRI 5.11.2?  If so, I'll upload and attach it to that release, and then we can refer people to that to try it.

Thanks.

CyberFox001, 2025-03-03:
No, I finally build the 5.10, the latest stable version of JRMI.

I think it's best to test on new thing at a time. :)

bobjacobsen, 2025-03-03:
I've renamed the file to jmri.5.10.flatpak and attached it to the JMRI 5.10 release. It can be downloaded from:

https://github.com/JMRI/JMRI/releases/download/v5.10/jmri.5.10.flatpak

Should we tell the jmri-developers list about it and ask them to try it?  Or the jmriusers list?

danielb987, 2025-03-03:
I would start with the JMRI developers list.

Daniel

bobjacobsen, 2025-03-03:
Sent to jmri-developers:  https://jmri-developers.groups.io/g/jmri/message/10859

CyberFox001, 2025-03-15:
Do we go next step ?

danielb987, 2025-03-16:
I'm fine with whatever you and @bobjacobsen decides.

Maybe the JMRI users list has more users interested in testing JMRI as flatpack.

CyberFox001, 2025-03-19:
On my side, I see no problem to announce the flatpak experiment on the user mailing list.

What do you think @bobjacobsen ?

bobjacobsen, 2025-03-19:
I think it would be fine to ask jmriusers about this,

CyberFox001, 2025-04-03:
I think I'm a little bit too new in the JMRI community to do the announce.

bobjacobsen, 2025-04-04:
I've posted it to jmriusers:

https://groups.io/g/jmriusers/message/241083

CyberFox001, 2025-09-13:
Hello. I come back about this package of JMRI.

So, from what I have read in the mailing list, and what I have tested, the main problems meets by users are:
- Varying support of Flatpak itself by the different GNU/Linux distribution
- One user tried to use the Flatpak package content without the container system of Flatpak
- The Flatpak package is only for Intel/AMD arch for now

On Linux Mint (tested on version 21 and 22), support for Flatpak is installed by default. With the Mint Software Center, you can install a Flatpak package from a repository (Flathub) but if you download it manually and double-click on the file, it don't know what to do with it. It's a bug from the Mint Software Center (I plan to create a report on it soon).

On Fedora Linux (tested on v. 42), Flatpak is installed by default and you can install a Flatpak package from a repo and from a manually downloaded file. Tested with the Gnome and KDE version.

On Ubuntu (tested 24.10 and 25.04), Flatpak is not installed by default. It's a choice from Canonical, it's editor. But, starting from version 25.04, you can install the "Flatpak Engine" from the Ubuntu Software Centre. For the previous version, you need to use the command line for this step. Sadly, Flatpak would no be integrated in the Ubuntu Software Center but use the KDE software manager instead.

On Debian (tested version 13), Flatpak support is not installed by default but it can easily be added. After that, you can install a Flatpak package from a repo or from a file manually downloaded.

Of course, before installing from a manually downloaded file, it's needed to add the Flathub repository. The Java runtime is distributed by Flathub.

The ARM edition also need to be done.

## tl;dr ##

If we distribute the JMRI Flatpak package via a repository like Flathub, it should work for most of the distributions.

If we distribute the JMRI Flatpak package manually:

- Linux Mint users will need to use command line to install, until the bug on the Software Center is fixed
- Ubuntu and Debian users need to add Flatpak first, then can install JMRI
- Fedora, it work

And I need to make an ARM version.