JMRI/JMRI #11695: SendAccPacket.py Script error

billmasters1971, 2022-12-29:
When I run jython script SendAccPacket.py right out of the box it generates an error in JMRI System Console:
ERROR - Unable to execute script. [AWT-EventQueue-0]
javax.script.ScriptException: SyntaxError: no viable alternative at input 'b2' in <script> at line number 37 at column number 12

Caused by:   File "<script>", line 37
    sendXOR([b1 b2 b3 b4 b5])

SyntaxError: no viable alternative at input 'b2'

devel-bobm, 2022-12-29:
I suspect that there are some "commas" are missing in that line.  It should be something more like:

`sendXOR([b1, b2, b3, b4, b5])`

devel-bobm, 2022-12-29:
FWIW, review of the code history shows that the missing commas have been absent since the code was first added to JMRI in April 2010.  Perhaps the Python (Jython) interpreter considered it acceptable code in the past, but current JMRI Python (Jython) interpreter version does not.  Or perhaps you are simply the first person to report this bug.

billmasters1971, 2022-12-29:
Thanks, I figured out the commas and got it past that point, but now the script returns a new error.  I've dome nothing but added the cammas:  sendXOR([b1, b2, b3, b4, b5]) and and edited the Adr, Cv and Val variables with my own values:
Adr = 2039
Cv = 137
Val = 10
New log from Console is:

`2022-12-29 05:20:21,735 jython.RunJythonScript                ERROR - Unable to execute script. [AWT-EventQueue-0]
javax.script.ScriptException: TypeError: sendPacket(): 1st arg can't be coerced to byte[] in <script> at line number 37
	at org.python.jsr223.PyScriptEngine.scriptException(PyScriptEngine.java:222)
	at org.python.jsr223.PyScriptEngine.eval(PyScriptEngine.java:59)
	at org.python.jsr223.PyScriptEngine.eval(PyScriptEngine.java:64)
	at java.scripting/javax.script.AbstractScriptEngine.eval(Unknown Source)
	at jmri.script.JmriScriptEngineManager.eval(JmriScriptEngineManager.java:360)
	at jmri.script.JmriScriptEngineManager.eval(JmriScriptEngineManager.java:297)
	at jmri.jmrit.jython.RunJythonScript.invoke(RunJythonScript.java:111)
	at jmri.jmrit.jython.RunJythonScript.actionPerformed(RunJythonScript.java:84)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton.doClick(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI.doClick(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(Unknown Source)
	at java.desktop/java.awt.Component.processMouseEvent(Unknown Source)
	at java.desktop/javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.Component.processEvent(Unknown Source)
	at java.desktop/java.awt.Container.processEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)
Caused by: Traceback (most recent call last):
  File "<script>", line 37, in <module>
  File "<script>", line 18, in sendXOR
  File "<script>", line 10, in send
TypeError: sendPacket(): 1st arg can't be coerced to byte[]

	at org.python.core.Py.TypeError(Py.java:236)
	at org.python.core.PyReflectedFunction.throwError(PyReflectedFunction.java:213)
	at org.python.core.PyReflectedFunction.throwBadArgError(PyReflectedFunction.java:316)
	at org.python.core.PyReflectedFunction.throwError(PyReflectedFunction.java:325)
	at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:171)
	at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:208)
	at org.python.core.PyObject.__call__(PyObject.java:494)
	at org.python.core.PyObject.__call__(PyObject.java:498)
	at org.python.core.PyMethod.__call__(PyMethod.java:156)
	at org.python.pycode._pyx2.send$1(<script>:10)
	at org.python.pycode._pyx2.call_function(<script>)
	at org.python.core.PyTableCode.call(PyTableCode.java:173)
	at org.python.core.PyBaseCode.call(PyBaseCode.java:134)
	at org.python.core.PyFunction.__call__(PyFunction.java:416)
	at org.python.pycode._pyx2.sendXOR$2(<script>:18)
	at org.python.pycode._pyx2.call_function(<script>)
	at org.python.core.PyTableCode.call(PyTableCode.java:173)
	at org.python.core.PyBaseCode.call(PyBaseCode.java:134)
	at org.python.core.PyFunction.__call__(PyFunction.java:416)
	at org.python.pycode._pyx2.f$0(<script>:37)
	at org.python.pycode._pyx2.call_function(<script>)
	at org.python.core.PyTableCode.call(PyTableCode.java:173)
	at org.python.core.PyCode.call(PyCode.java:18)
	at org.python.core.Py.runCode(Py.java:1687)
	at org.python.core.__builtin__.eval(__builtin__.java:497)
	at org.python.core.__builtin__.eval(__builtin__.java:501)
	at org.python.util.PythonInterpreter.eval(PythonInterpreter.java:255)
	at org.python.jsr223.PyScriptEngine.eval(PyScriptEngine.java:57)
	... 43 more
`

I have an alternative for what I'm trying to do using AutoHotkey, but if I could get this script to work it would be much smoother.

devel-bobm, 2022-12-29:
for my next _guess_:

`sendXOR(bytearray[b1 b2 b3 b4 b5])`

Perhaps that will give you some more progress.  

I am not sure where `bytearray` method is defined, so I cannot guess whether it will require an "import".  I borrowed this concept from `jython/LCC/SyncMemories.py` and `jython/TelnetTrain/py`.  Perhaps one of these will provide some sort of help, if needed.

dsand47, 2022-12-29:
The error is actually on line 10.  The most recent call is listed last. 

A "bytearray" is a Python sequence (list) that contains integer values from 0 to 255.

Jython provides a "jarray" method to convert Python lists into Java arrays.  This available using "import jarray".

There are several examples where a list of named beans is converted to a Java array.
"self.setInputs(jarray.array([self.s1, self.s2, self.s3], jmri.NamedBean))"

There are a number of options for the second parameter in addition to "jmri.NamedBean".  See https://www.jython.org/jython-old-sites/archive/21/docs/jarray.html.  For this script, the 'b' option is used.

In theory, the following should work:

    dcc.sendPacket(jarray.array(bytearray(pkt), 'b'), 1)

However, Jython 2.7 changed the behavior of the byte conversion.  It appears that values greater than 127 create errors.

This change solves the sendPacket conversion issues:

def send(pkt):
    pklist = []
    for p in pkt:
        if p < 128:
            pklist.append(p)
        else:
            pklist.append(p - 256)
    dcc.sendPacket(jarray.array(pklist, 'b'), 1)

The result using address 765 (instead of 2041) and the LocoNet simulator:

ED 0B 7F 50 07 5F 58 6E 02 00 5A]  Send packet immediate: 5 bytes, repeat count 0(80)
            DHI=0x07, IM1=0x5F, IM2=0x58, IM3=0x6E, IM4=0x02, IM5=0x00
            packet: DF D8 EE 02 00 .

The other problem is that the packet calculations are not valid.  In the example, the value for "b1" using 2041 is 383.  

Dave Sand

billmasters1971, 2022-12-30:
Thanks for taking a look.  I managed to get it to the point where I discovered that B1 was greater than 255 but with my limited Jython/Python knowledge there's no way I'll get past that point.  I'm more of a Xojo programmer.  I do have a less efficient workaround to get me going so I'll just check back to see if the script works in future releases and if it does I will most definitely use it.  Surprised nobody has caught it before me because that is such a useful script.

closed 2024-05-06 by github-actions[bot]