JMRI/JMRI #11696: request:  Add Reporter to MQTT

nigelcliffe, 2023-01-01:
As far as I can tell, the MQTT support does not yet include Reporters.  (I am running JMRI 5.3.1). 

I am working with someone on building some RailCom detectors.  They currently work correctly over CBUS (using the MERG format for data, giving loco ID and orientation), but we're considering modifying the detector hardware to also work via MQTT, and thus giving us the option to install the detectors in locations where extending the wired network would be awkward.  

For us, the simplest option may be support for Reporters in the MQTT part of JMRI.   So, adding this could be useful, hence this request. 

Our alternative (which probably works) is to subscribe to a topic using LogixNG and deal with things from there.    

Thanks,

Nigel Cliffe

bobjacobsen, 2023-01-01:
It's been a while since I've worked on Reporters. Things have changed a bit, particularly around the connection between reporters and tags.

What's your thinking about the MQTT content and how it should map to the Reporter's report value? Is that just a one-to-one passthrough of a string, or something else?

nigelcliffe, 2023-01-01:
The simplest for us would be the same payload we send when using CBUS, which is several bytes of data (some are null's), which encodes a RailCom identity and the Orientation of the loco.   
Then, in the reporter table, being able to select the CBUS reporter type of "CANRC522/CANRCOM", will automatically decode the byte payload, dropping a RailCom ID into the "correct" place, appearing in IDTags, etc, and all the existing features in LogixNG to handle RailCom reports should work with no changes.  

However, I can see others wanting to use a different payload, and thus having an option in the reporter table (as is the case with CBUS reporter) to select the parsing of the payload is likely to be useful:   even if the initial offering is just two cases: the CANRCOM parse (as above) and "no parsing, raw string".   

- Nigel

bobjacobsen, 2023-01-03:
> The simplest for us would be the same payload we send when using CBUS

That is turning out to be remarkably hard.

The CBUS reporter code has lots of tentacles intertwined with various over parts of the CBUS code, and I don’t see how to disentangle it.

I _think_ (in the sense that I’m not yet sure) that I can create an MQTT reporter that takes an arbitrary string input and makes a DefaultIdTag from it.  The uncertainty is because it’s clear I don’t understand how IdTags are interconnected with Reporters.  That seems to vary from one part of the code to another.

More to follow.

nigelcliffe, 2023-01-03:
OK, if its turning out to be hard to do the CBUS way, then the basic requirements are:

Need to transfer a message containing both loco address (number) and orientation (three states, currently handled with digits 0,1,2 for “unknown”, “left”, “right”).    We can format message to match what’s needed.   We could have two messages (address, orientation), but that seems very messy.
When a detector looses a loco, it will send an empty report to indicate the section is empty.

Once inside JMRI, the loco address goes to “ID tags”, etc..  as per any other reporter.   Orientation is currently handled by scripts looking at the reporter message (LogixNG) to extract and do useful things to it.

  *   Nigel
  *

bobjacobsen, 2023-01-04:
Thanks.  I've got the basic code in place, now need to work on parsing.

I think one message is the right approach.  I can do loco number, direction e.g.

```
123 right
456 unknown
```

but should there also be an entry/exits term in there?

```
123 right entry
456 unknown exits
```

nigelcliffe, 2023-01-04:
> From "Bob Jacobsen" ***@***.***>
> To "JMRI/JMRI" ***@***.***>
> Cc "nigelcliffe" ***@***.***>; "Author"
> Date 04/01/2023 16:18:05
> Subject Re: [JMRI/JMRI] request: Add Reporter to MQTT (Issue #11696)

We could add entry/exit message to the device sending the MQTT messages 
if they are required.   I think the entry/exit side comes out of 
Digitrax' Transponding.
One approach is to use entry and exit messages.  The other would be an 
inferred "exit" which would happen if the reporter says "null" (empty 
string) when it is now empty, and an inferred "entry" when a report 
arrives with a value.

One thing which does happen with our device is a timing issue.   The 
initial read of the RailCom identity is faster than establishing 
orientation.   So, our current CBUS device will send two messages when a 
RailCom decoder enters the detection area, the first will be "0123 0" 
(loco 123 orientation unknown) and the next is a second or so later as 
"0123 1"  (loco 123 orientation right).   We could change that so the 
device waits until it has found the orientation (or timed out and gone 
without orientation).

I don't know if you want to chat about this ?   I'm currently at 
computer and could fire up zoom.

regards

Nigel

closed 2023-01-06 by danielb987