JMRI/JMRI #11729: html pages - change input name="button" to css

icklesteve, 2023-01-09:
Screenreaders and many other software packages do not understand the complete sentence when inputs are being used to style text mid-sentence.

eg. html "Save to Roster Button" towards end of
https://www.jmri.org/help/en/manual/DecoderPro/Adv_FunctionLabel.shtml

`make sure you click on the <input name="button" id="button" value="Save to Roster" type="submit"> to save`

When using the MS Windows built-in screen reader [MS Narrator](https://support.microsoft.com/en-us/windows/complete-guide-to-narrator-e4397a0d-ef4f-b386-d8ae-c172f109bdb1#WindowsVersion=Windows_11), text within inputs will not be read out with the accompanying text, it's focusing on output to user, not input.

In the eg. both internal Narrator within Edge, and Win11 Narrator in Google Chrome reads out 

> make sure you click on the to save

There are about 140 of these currently in the support files.

I'm proposing conversion to a span with the `textbutton` class as suggested in https://github.com/JMRI/JMRI/pull/11584#issuecomment-1311801199 eg.

`make sure you click on the <span class="textbutton">Save to Roster</span> button `

danielb987, 2023-01-09:
I agree.

danielb987, 2023-01-09:
@silverailscolo 
I mention you in this issue since it's related to https://github.com/JMRI/JMRI/pull/11584.

silverailscolo, 2023-01-10:
Agreed

bobjacobsen, 2023-03-29:
If the change is to replace
```
<input name="button" id="button" value="Some Text" type="submit">
```
with exactly 
```
<span class="textbutton">Some Text</span>
```
I'd be happy to go through and make those changes. There seem to be 26 occurrences of the basic pattern, plus three more with id="button2" et al.

closed 2023-06-08 by icklesteve