JMRI/JMRI #11740: SRCP Server problems

bianchina3, 2023-01-13:
Hi.
I'm using the official SRCP server from http://srcpd.sourceforge.net/ ver. 2.1.5 and I've encountered some protocol problems with JMRI.
browsing the web I saw that other users have had the same problems including at least two do not allow the use of JMRI.

1) JMRI expects the welcome message from Server to be timestamped, which is not coherent with SRCP specifications.
    SRCP welcome message must be composed of 1 up to several key/value pairs. No timestamp.

2) JMRI expects the 2 ascii codes "\r" and "\n" as end of line, which is not coherent with SRCP specifications.
    Specified end-of-line is "\n" as last character.

bobjacobsen, 2023-01-13:
Could you say a bit more about how you determined those two problems please?

From just reading the code (which could be giving me a wrong impression), the input parser should be accepting \n or \r or \n\r.

bianchina3, 2023-01-14:
Ok. 

**************************************************************************************        
test for the demonstration of problem number 1
SRCPD software version 2.1.5 original. listen port 4303
JMRI say parser error on SRCP welcome string

 
dummy@bartonote4:/opt/JMRI$ ./DecoderPro
2023-01-14 10:05:27,487 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2023-01-14 10:05:27,497 util.Log4JUtil                        INFO  - This log is appended to file: /home/dummy/.jmri/log/messages.log [main]
2023-01-14 10:05:27,498 util.Log4JUtil                        INFO  - This log is stored in file: /home/dummy/.jmri/log/session.log [main]
2023-01-14 10:05:27,581 apps.AppsBase                         INFO  - DecoderPro version 5.2+R760b98537f starts under Java 11.0.16 on Linux amd64 v5.10.0-20-amd64 at Sat Jan 14 10:05:27 CET 2023 [main]
2023-01-14 10:05:27,974 gui3.Apps3                            INFO  - Starting with profile My_JMRI_Railroad.3e9cb7fd [main]
2023-01-14 10:05:28,065 apps.AppsBase                         INFO  - No pre-existing config file found, searched for '/home/dummy/.jmri/My_JMRI_Railroad.jmri/ProfileConfig.xml' [main]
2023-01-14 10:05:28,065 util.FileUtilSupport                  INFO  - File path program: is /opt/JMRI/ [main]
2023-01-14 10:05:28,066 util.FileUtilSupport                  INFO  - File path preference: is /home/dummy/.jmri/My_JMRI_Railroad.jmri/ [main]
2023-01-14 10:05:28,066 util.FileUtilSupport                  INFO  - File path profile: is /home/dummy/.jmri/My_JMRI_Railroad.jmri/ [main]
2023-01-14 10:05:28,066 util.FileUtilSupport                  INFO  - File path settings: is /home/dummy/.jmri/ [main]
2023-01-14 10:05:28,066 util.FileUtilSupport                  INFO  - File path home: is /home/dummy/ [main]
2023-01-14 10:05:28,067 util.FileUtilSupport                  INFO  - File path scripts: is /opt/JMRI/jython/ [main]
2023-01-14 10:05:28,450 node.NodeIdentity                     INFO  - Using 07386075-bede-4ebe-b9d9-ddc32dee84b2 as the JMRI storage identity for profile id 3e9cb7fd [main]
2023-01-14 10:05:50,481 jmrix.AbstractMRTrafficController     ERROR - run: Exception: jmri.jmrix.srcp.parser.ParseException: Encountered " <SRCP> "srcp "" at line 1, column 1.
Was expecting one of:
    <ZEROADDR> ...
    <NONZEROADDR> ...
    <TIMESTAMP> ...
     in class jmri.jmrix.srcp.SRCPTrafficController [srcp.SRCPTrafficController Receive thread]
jmri.jmrix.srcp.parser.ParseException: Encountered " <SRCP> "srcp "" at line 1, column 1.
Was expecting one of:
    <ZEROADDR> ...
    <NONZEROADDR> ...
    <TIMESTAMP> ...
    
        at jmri.jmrix.srcp.parser.SRCPClientParser.generateParseException(SRCPClientParser.java:1506)
        at jmri.jmrix.srcp.parser.SRCPClientParser.jj_consume_token(SRCPClientParser.java:1374)
        at jmri.jmrix.srcp.parser.SRCPClientParser.timestamp(SRCPClientParser.java:100)
        at jmri.jmrix.srcp.parser.SRCPClientParser.handshakeresponse(SRCPClientParser.java:34)
        at jmri.jmrix.srcp.SRCPTrafficController.receiveLoop(SRCPTrafficController.java:88)
        at jmri.jmrix.AbstractMRTrafficController$2.run(AbstractMRTrafficController.java:803)
        at java.base/java.lang.Thread.run(Thread.java:829)

        
TCP packet sniffed
note that SRCPD send The "SRCP welcome string" with '\n' (0a) terminator and without timestap as SRCP specification:

10:27:48.230342 IP localhost.4303 > localhost.54246: Flags [P.], seq 1:43, ack 1, win 64, options [nop,nop,TS val 3426351358 ecr 3426351357], length 42
        0x0000:  0000 0000 0000 0000 0000 0000 0800 4500  ..............E.
        0x0010:  005e afbe 4000 4006 8cd9 7f00 0001 7f00  .^..@.@.........
        0x0020:  0001 10cf d3e6 ceae a6e0 d49e f916 8018  ................
        0x0030:  0040 fe52 0000 0101 080a cc39 f8fe cc39  .@.R.......9...9
        0x0040:  f8fd 7372 6370 6420 5632 2e31 2e35 3b20  ..srcpd.V2.1.5;.
        0x0050:  5352 4350 2030 2e38 2e34 3b20 5352 4350  SRCP.0.8.4;.SRCP
        0x0060:  4f54 4845 5220 302e 382e 330a            OTHER.0.8.3.

**************************************************************************************        
test for the demonstration of problem number 2
SRCPD software version 2.1.5 hacked for no send "SRCP welcome string". listen port 4303
JMRI doesn't get upset about the missing string and just keeps running but say "Timeout on reply to message".

dummy@bartonote4:/opt/JMRI$ ./DecoderPro
2023-01-14 11:04:23,578 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2023-01-14 11:04:23,586 util.Log4JUtil                        INFO  - This log is appended to file: /home/dummy/.jmri/log/messages.log [main]
2023-01-14 11:04:23,588 util.Log4JUtil                        INFO  - This log is stored in file: /home/dummy/.jmri/log/session.log [main]
2023-01-14 11:04:23,671 apps.AppsBase                         INFO  - DecoderPro version 5.2+R760b98537f starts under Java 11.0.16 on Linux amd64 v5.10.0-20-amd64 at Sat Jan 14 11:04:23 CET 2023 [main]
2023-01-14 11:04:24,075 gui3.Apps3                            INFO  - Starting with profile My_JMRI_Railroad.3e9cb7fd [main]
2023-01-14 11:04:24,185 node.NodeIdentity                     INFO  - Using 07386075-bede-4ebe-b9d9-ddc32dee84b2 as the JMRI storage identity for profile id 3e9cb7fd [AWT-EventQueue-0]
2023-01-14 11:04:24,843 util.FileUtilSupport                  INFO  - File path program: is /opt/JMRI/ [main]
[... cut ...]
        0x0020:  0001 10cf ba1a 866d d39e 7b53 93c0 8018  .......m..{S....
        0x0030:  0040 fe4d 0000 0101 080a cc76 f23a cc76  .@.M.......v.:.v
        0x0040:  f23a 3136 3733 3639 3234 3634 2e31 3934  .:1673692464.194
        0x0050:  2032 3031 204f 4b20 5052 4f54 4f43 4f4c  .201.OK.PROTOCOL
        0x0060:  2053 5243 500a 0d                        .SRCP..

        
The warning "Timeout on reply to message" are gone and JMRI work well.
It seems that JMRI only accepts SRCP response strings ending with "\n\r" (0a0d).

bobjacobsen, 2023-01-14:
Unfortunately, I'm neither a SRCP user nor a JavaCC programmer, so I can only do a little on this.

The CRLF issue seems to be due to how `EOL` is defined in `java/src/jmri/jmrix/srcp/SRCPClientParser.jjt`.  

Note that the tests in `java/test/jmri/jmrix/srcp/parser/SRCPClientParserTest.java` end their input lines with `\n\r`.  If the `\r` is removed, the tests fail either because the parser doesn't trigger expected errors or with a message like:
```
  JUnit Jupiter:SRCPClientParserTest:testServerDescriptionResponse()
    MethodSource [className = 'jmri.jmrix.srcp.parser.SRCPClientParserTest', methodName = 'testServerDescriptionResponse', methodParameterTypes = '']
    => jmri.jmrix.srcp.parser.TokenMgrError: Lexical error at line 2, column 0.  Encountered: <EOF> after : ""
       jmri.jmrix.srcp.parser.SRCPClientParserTokenManager.getNextToken(SRCPClientParserTokenManager.java:792)
```

From this I conclude that if we can find a way to fix the parser, we'll be able to test that. The JavaCC source for the parser (itself to be parsed into Java code) can be found in `java/src/jmri/jmrix/srcp/SRCPClientParser.jjt`.  The relevant part seems to be 
```
<DEFAULT,IN_REASON>
TOKEN :
{
  < EOL :(["\n","\r"]){2} >:DEFAULT
}
```

I haven't been able to figure out how to change that to avoid the no-\r test failures, unfortunately.  Perhaps somebody who understands JavaCC better can do that.  I've tried:

```
{
  < EOL :(["\n"){2} >:DEFAULT
}
```

```
{
  < EOL :(["\n"]){1} >:DEFAULT
}
```

Both of which don't seem to make any difference (which is unexpected) and

```
{
  < EOL : "\n" >:DEFAULT
}
```
which fires a parse error because NEWLINE also defines the same thing later in the file.

The problem with the timestamp might be due to 
```
% grep -nr TIMESTAMP java/src/jmri/jmrix/srcp
java/src/jmri/jmrix/srcp/SRCPClientParser.jjt:68:| < TIMESTAMP: (["0"-"9"])+((".")(["0"-"9"])+)? >
java/src/jmri/jmrix/srcp/SRCPClientParser.jjt:147:   (t=<TIMESTAMP> | t=<NONZEROADDR> | t=<ZEROADDR> )
java/src/jmri/jmrix/srcp/SRCPReply.java:64:        // split into 3 parts {TIMESTAMP, ResponseCode, Rest}
```

but, as in the end-line case, I don't know how to change that parsing so that it works.

bianchina3, 2023-01-14:
We just have to wait for someone who knows javacc well and can fix these bugs.

bobjacobsen, 2023-01-14:
Is JMRI sending \n\r at the end of messages a problem?

bianchina3, 2023-01-15:
For SRCPD messages ending with "\n\r" are not a problem. He accepts them but the fact remains that these messages are out of SRCP specification.

SRCP 0.8.3 specification in paragraph 2.5 says:
The end of line always consists of the character #10(\n, LF). A prefixed #13(\r, CR) is valid. The
character #13 is ignored. A single line including the end of the line MUST NOT exceed over a length of
1000 characters. Information sent by the server is subject to the same terms. The character #13 CAN be
sent immediately before #10.

Thus specified end-of-line is "\n" as last character, with a potential but non mandatory "\r" before.

danielb987, 2023-02-20:
@bobjacobsen 
> Is JMRI sending \n\r at the end of messages a problem?

I'm pretty sure Windows is using \r\n, not \n\r.

pin88303, 2023-04-15:
Hello all, I just have the issue problem. I just wrote a message on the group for it with attached files:
`

    2023-04-14 21:29:59,796 jmrix.AbstractMRTrafficController WARN - Exception java net: [srcp.SRCPTrafficController Receive thread]
    jmri.jmrix.srcp.parser.ParseException: Encountered " <SRCP> "srcp "" at line 1, column 1.
    Was expecting one of:
    <ZEROADDR> ...
    <NONZEROADDR> ...
    <TIMESTAMP> ..

Thx, Marc

`
I'm working on debian11, last version of srcp and last version of JMRI

pin88303, 2023-04-16:
Hello,
I tried different tests to modify:

 `  < EOL :(["\n","\r"]){2} >:DEFAULT
}
<DEFAULT>
SKIP:
{
     <RETURN: "\r">
   | <NEWLINE: "\n">
   | <SPACE: " ">
}` 
from line 48 to 56 in the SRCPClientParser.jjt file but without success, I don't have any knowledges in javacc ....

This is the way i hacked SRCP in file srcp-server.h to connect JMRI and SRCP, waiting a better solution from an expert in javacc:
`/* SRCP server welcome message */
static const char WELCOME_MSG[] =
/*"srcpd V" VERSION "; SRCP 0.8.4; SRCPOTHER 0.8.3\n";*/
"\n";`

I still have the same warning's timeout problems as mentioned above which is not critical. I still investigate.

Hope it could help some people.

SRCP 2.1.6 and Building JMRI 5.3.5-SNAPSHOT on debian11

Marc

pabender, 2023-05-07:
@pin88303 @bianchina3 could you check to see if the code in #12007 resolves the issues.  That PR should fix the parsing issues, but I haven't had a chance to get srcpd working, so I don't know how it behaves with that server.

The key value pairs on the welcome message was a big miss when the parser was written.

Expecting the timestamp on the welcome message is a matter of ambiguity in the standard.  Section 2.6 on replies indicates that all replies must be preceded by a timestamp.   The welcome message is just another message in the protocol.

The way I've re-written this, the timestamp on the welcome message is now optional.  It's still required for everything else.

Incidentally, on the end of line, it was written into the parser so that the parser itself doesn't care what order the  \n and \r occur in.

bobjacobsen, 2023-05-08:
Would it help if I built a set of installers with #12007's changes in them?  Happy to do that if it'd be useful for testing.

danielb987, 2023-05-08:
> Would it help if I built a set of installers with #12007's changes in them? Happy to do that if it'd be useful for testing.

I guess so, but I don't know.

bobjacobsen, 2023-05-08:
There are installers available for the current #12007 branch at:

https://builds.jmri.org/jenkins/job/testreleases/job/srcpeol/52/

pabender, 2023-05-08:
> Hello again,
> 
> Here are the log files under windows 11
> 
> Sincerely
> 
> Marc [session.log](https://github.com/JMRI/JMRI/files/11422739/session.log) [messages.log](https://github.com/JMRI/JMRI/files/11422740/messages.log)

Those logs actually do show a parsing error.

It would be helpful to see the full input here, but that's not something we can log.  It's interesting that whatever you're seeing is only happening on Windows.

pabender, 2023-05-09:
@pin88303 how do you have srcpd configured?  What hardware interface are you using?  

The only real hardware I have that srcpd supports is a Lenz LI100.  When I connect that, srcpd sees the device, and tries interacts with the command station, but it stalls before it will allow connections.  I'm not going to debug the code involved further to figure out what's wrong in srcpd itself.

If I configure srcpd to use a loopback connection, the JMRI SRCP client connects just fine, but the server seems to only respond to most commands when a command is sent in error.  It doesn't ever respond to most real commands (I got it to respond to the description of the buses, but i get no response when I try to do anything with device in a supported device group on one of those buses).

I did setup wireshark so I could trace the connection in this case, it didn't show me anything I wasn't seeing in the JMRI logs.

The test with the loopback configuration proves that the changes I made in PR #12007 actually allow connections to be made to srcpd.  There may still be other parsing issues, but until I get further information, I have no way to replicate that.

closed 2023-05-09 by danielb987

pin88303, 2023-05-09:
Hello, im' really not a pro, but i will try to give informations as much as i can ;)

Here is my config:

Debian 11
Central connected on:
" Detected FT232RL
[ 4486.479494] usb 8-2: FTDI USB Serial Device converter now attached to ttyUSB0"

But i did a lot of tests on my side and i'm pretty sure that problems do not come from jmri.
I think i have serial mistake with my central and pc, which is internal of my configuration.
I will first resolve this and come back.

Thx to the jmri'team for this nice work :)