JMRI/JMRI #11743: help/en/Logging.shtml references "Program Directory" and "Preferences" directory

newHeiko, 2023-01-13:
Hi everyone,

The file Logging.shtml from the help has the following: https://github.com/JMRI/JMRI/blob/7ca3b1ca43feea7b4500fb9a254816cf849a6984/help/en/html/doc/Technical/Logging.shtml#L85

talking about a "Program Directory" and a "Preferences" directory which I shall find through Help->File Locations in the menu.

However, my 5.2/Linux/Azul JDK 11.0.17 setup shows the following window when I click Help->File Locations from the Roster window (DecoderPro: All Entries main window):
[image]

I'm guessing "Program Directory" is called "Program Location" there, but what's "Preferences" directory supposed to mean?

"Settings Location"?
"User Files Location"?
"Profile Location"?

I suggest matching the wording between the website and the display in Help->File Locations.

Thanks,
Heiko

dsand47, 2023-01-13:
Heiko,

That reference is the "settings" location.  

Dave Sand

However, my 5.2/Linux/Azul JDK 11.0.17 setup shows the following window when I click Help->File Locations from the Roster window (DecoderPro: All Entries main window): image <https://user-images.githubusercontent.com/35038740/212384120-1a4e8605-e48c-4f1f-8668-68310bc32bb5.png>

newHeiko, 2023-01-13:
Thanks Dave,

I've created PR #11744 to re-word the text on that page.

Heiko

PR #11744 (2023-01-14): Match directory names to output of Help->File Locations
Fixes #11743
commit 763edf95 (2023-01-14): Match directory names to output of Help->File Locations

Fixes #11743
closed 2023-01-14 by danielb987