JMRI/JMRI #11747: Script copies Turnout State changes from one connection to another connection

Cliff-Anderson, 2023-01-14:
In response to a suggestion from someone on the JMRI Users Group, this script is being submitted for your consideration and possible inclusion in the JMRI downloads.  
  
While rare, some layouts have the need to install two active connections with throttles on one bus and turnouts on another.  
  
The earliest occurrence of the problem that came to my attention was in the message thread "Can you turn LCC events into DCC Accessory Commands" https://groups.io/g/jmriusers/topic/19534425#139700  
  
At that time, with a lot of help from @bobjacobsen and @dsand47 a script was uploaded to the groups.io "jmriusers/files/Script examples/Master and Slave Turnout Control" folder.  
   
Recently, a slightly different request for help with the message thread "Multiple Openlcb Conections" https://groups.io/g/jmriusers/topic/19534085#163660  
  
As a result of that request and the suggestion from a group member, an updated version of the script "TurnoutsMasterSlave.py'" has been verified with versions 5.2 and 5.3.1 and is included here.    
[TurnoutsMasterSlave.zip](https://github.com/JMRI/JMRI/files/10416360/TurnoutsMasterSlave.zip)

The user is requested to modify lines 70 and 71 to match their local DCC bus requirements, but the script is otherwise agnostic as to DCC hardware.  Two-way communication of state changes and related confusion is NOT considered.

A very simplistic PanelPro file is also included here for you to check out with the "as is" script to copy turnouts changes on a simulated LocoNet bus to turnouts defined on a simulated NCE bus.  
[LocoNet to NCE test.zip](https://github.com/JMRI/JMRI/files/10416379/LocoNet.to.NCE.test.zip) and was used for my testing.

bobjacobsen, 2023-01-14:
Thanks!  See PR #11749 which will merge this.

danielb987, 2023-01-16:
Closing this since the script is added to JMRI in PR #11749 which is merged.

closed 2023-01-16 by danielb987