JMRI/JMRI #11765: Analog clock only shows integral rate

devel-bobm, 2023-01-27:
The Analog Clock window shows the fast clock's rate as a whole number even though the fast clock supports fractional rates and happens to be running at a fractional rate.

https://github.com/JMRI/JMRI/blob/55481d5e1646f897ca5e591f70a880708247eaa3/java/src/jmri/jmrit/analogclock/AnalogClockFrame.java#L288

This can be seen when using the default fast clock when the rate is not a whole number.

Also note that the the displayed rate is _only_ updated when the minute hand moves.  This can make it difficult to know that the rate has actually been changed.  This is because the only thing that gets updated by the property change event handler:

https://github.com/JMRI/JMRI/blob/55481d5e1646f897ca5e591f70a880708247eaa3/java/src/jmri/jmrit/analogclock/AnalogClockFrame.java#L292-L300

is Running/Paused status:
https://github.com/JMRI/JMRI/blob/55481d5e1646f897ca5e591f70a880708247eaa3/java/src/jmri/jmrit/analogclock/AnalogClockFrame.java#LL301-L307

devel-bobm, 2023-01-27:
... ia Running/Paused status:
https://github.com/JMRI/JMRI/blob/55481d5e1646f897ca5e591f70a880708247eaa3/java/src/jmri/jmrit/analogclock/AnalogClockFrame.java#L301-L307

devel-bobm, 2023-01-27:
Apparantly the clock implementation does not fire a property change event when the rate is changed...

bobjacobsen, 2023-01-27:
PR #11766 fixes the fractional rate.  It'll also update the rate as soon as the underlying clock fires propertyChange on a rate change. Firing an event is not included here.

bobjacobsen, 2023-01-27:
Rate change is not firing an event due to an explicit check in SimpleTimebase:
```
        if (internalMaster) {
            firePropertyChange("rate", oldFactor, factor); // old, then new
        }
```
This seems odd, as I would have thought that the internal clock would have `internalMaster` set true.

devel-bobm, 2023-01-28:
FWIW - my experiments were done with a LocoNet Simulator config profile.

closed 2023-01-28 by bobjacobsen