JMRI/JMRI #11781: JMRI 5.2 will not start

bpupnorth, 2023-02-02:
I have a Dell laptop running the latest Windows 10 updates. I installed JMRI 5.2 and it won't start, install test shows return code 1. I have tried two versions of Java 11, using Oracle at the moment. Installed to D:\JMRI and also to the default C: location. JMRI Version 5.0 runs fine.

Pugwash1, 2023-02-02:
Maybe a  32bit java on 64 bit OS.  From groups.io "if you want to use a 32bit java on a 64bit  windows OS you can try removing java from the path so it cant find the default and falls back on the registry, or add **_/32bit_** after **_LaunchJMRI.exe_** in the properties of the PanelPro start icon."
If you want to force a version then the java needs to be in the  **{jmri install directory}/jre**  as described here [https://www.jmri.org/install/WindowsNew.shtml](url).

bpupnorth, 2023-02-02:
Thanks for the reply. Here's the Java version, 64 bit.

C:\Users\Casey Jones>java -version
java version "11.0.17" 2022-10-18 LTS
Java(TM) SE Runtime Environment 18.9 (build 11.0.17+10-LTS-269)
Java HotSpot(TM) 64-Bit Server VM 18.9 (build 11.0.17+10-LTS-269, mixed mode)

5.0 runs fine with it.

Pugwash1, 2023-02-02:
You can try 
A -Find where java 11 was installed, this is unlikely to be the location in the path, that points to a directory that points to symlinks. Create a symlink in the JMRI directory called jre that points to final directory, which will have 11 in its name.
B - Try setting your path to not include the java paths, you can try in a command prompt, just set path=whatever minus the java paths. cd to where you installed jmri and run **LaunchJMRI /debug apps.PanelPro.PanelPro**

Pugwash1, 2023-02-02:
In a command prompt do **where java**
if it says this...
c:................\java\javapath\java.exe   then we have a problem.
it should say
c.................\jdk-11.0.17.....\bin

bpupnorth, 2023-02-03:
Thanks for this. Looks like I have the problem path. I guess that's a difference from JMRI 5.0. I'll reload everything and get the java path right and try again. I want to clean it all up and start fresh.

bpupnorth, 2023-02-03:
Pugwash1 - Thank you! I uninstalled and reinstalled everything with my JMRI on my D: drive. As you noted the Java path was different. I set my path to PATH=C:\Program Files\Java\jdk-11.0.17\bin;.......................; and all worked as it should. I pulled in my backup files and am back on track. Thanks again!

danielb987, 2023-02-04:
@bpupnorth 
Welcome to the JMRI project!
If this issue is resolved, it would be good to close it.

closed 2023-02-04 by bpupnorth