JMRI/JMRI #11793: Duplicate 'All Entries' in the roster groups

KenC57, 2023-02-09:
From a user (Morgan B) we captured the issue and Dave S identified it. Somehow a keypair is saved into a loco entry of the group 'All Entries' and when the roster index gets rebuilt, it shows as two entries of 'All Entries' in the list. Have not looked into how this can happen. Hoping someone familiar with the keypairs and roster groups might have a clue.

I think some sort of script might help users key up from however this happens. I've not found anything damaged by this entry other than it looks annoying.

KenC57, 2023-02-09:
[My_JMRI_Railroad.zip](https://github.com/JMRI/JMRI/files/10701747/My_JMRI_Railroad.zip)
This is a copy of the users profile that shows the problem. The loco in question is the PRR_4538_I1sa_QSI_4.xml file.

dsand47, 2023-02-10:
The duplicate `All Entries` group is caused by `Roster Group Table Association`.  It can also remove the duplicate.

When the `All Entries` group is selected in the dialog, all of the roster entries are unchecked even though the entries are active on the main roster page.  If one or more roster entries are then selected, the second entry is created and will show up the next time JMRI is started.

To resolve the problem, do the `Roster Group Table Association` action, select either `All Entries` from the drop down and uncheck any roster entries.  The extra `All Entries` will be gone after the next JMRI start.

KenC57, 2023-02-11:
It's too bad the column in the Roster Group Table are not sortable. Being able to click on the 'Include' wound make this a snap for someone with large rosters. Do you think it worth adding something to the help to point out how to remove the duplicates?

closed 2023-07-29 by bobjacobsen