JMRI/JMRI #11800: Problème avec la version V5.2

JoelDanieleDavid, 2023-02-13:
Issue Type: __Bug__

Good morning
I was under version V5.0 since its release.
I just downloaded and installed the V5.2 version without any problem.
I'm on Windows 10 and Java version 11 latest update.
I use Microsoft Defender Antivirus and Firewall.
I double-click on the JMRI V5.2 icon installed on the desktop, the software refuses to open. No error message.
I uninstall V5.2 and reinstall V5.0, everything works fine.
When I launch the "InstallTest" file, under V5.0, the results are ok. The same manipulation under V5.2 and the results are negative, it informs me of a problem with this version without further details.
I suddenly uninstalled the version in place and installed the intermediate test versions V5.1.X. It turns out that up to version 5.1.5 the software starts correctly, but from V5.1.6 up to V5.2 the software refuses to start. I also tried with V5.3.X test builds, unable to boot as well.
What could be the problem ?
Thank you in advance for your return.
Joel

DecoderPro 5.0+Rc441642522
Java HotSpot(TM) 64-Bit Server VM Oracle Corporation 11.0.17+10-LTS-269 mixed mode
Windows 10 10.0 amd64

bobjacobsen, 2023-02-14:
@Pugwash1 Could this be related to #11499?

Pugwash1, 2023-02-14:
@bobjacobsen  We ask java what version are you, it replies 11. Great... execute. The java found was an oracle java.exe version 11 but does not act like it. I cannot reproduce and I have 1.2 thru 24.0, so you'd think I would have the problem. I believe it's caused by bad registry keys, or bad paths. There is an ancient napi explorer java add in which has a version of 11 that confuses everything. 
If the installed java is linked or installed into the jmri/jre directory as per instructions that always works (providing javapath is not used, which maybe the oracle hell hole), removing all javas from the path works if the install added javasoft keys (the latest oracle doesn't load those by default, as do a lot of the others). 
Compounding this is that the install and launch are 32bit and I don't know if it understands the long path strings, so paths longer than 240 maybe iffy.

The first bit of diagnosing the problem is to run these commands.

FOR /F "usebackq delims==" %i IN (`where /F java`) DO dir "%~di%~pijava*"
reg query HKLM\SOFTWARE\JavaSoft /s
ECHO %PATH%

JoelDanieleDavid, 2023-02-14:
Hello to both of you. Thanks to your explanations, I found the solution. Thank you. Joel

danielb987, 2023-03-06:
@JoelDanieleDavid 
Can this Issue be closed?

JoelDanieleDavid, 2023-03-06:
Good morning.
Yes, for me the subject may be closed.
The answers obtained brought me the solution.
Thanks again.
Joel

closed 2023-03-06 by JoelDanieleDavid