JMRI/JMRI #11801: Not able to Add Blocks to Sections

Mack9000, 2023-02-14:
Issue Type: __Bug__
[profile.zip](https://github.com/JMRI/JMRI/files/10728698/profile.zip)
[session.log](https://github.com/JMRI/JMRI/files/10728699/session.log)
[systemInfo.txt](https://github.com/JMRI/JMRI/files/10728700/systemInfo.txt)

Issues with adding new sections.  No options to add blocks to new section. 

PanelPro 5.2+R760b98537f
OpenJDK Server VM Raspbian 11.0.16+8-post-Raspbian-1deb10u1 mixed mode
Linux 5.10.103-v7l+ arm

dsand47, 2023-02-14:
Adding blocks to a new section works for me.
[image]

Select a block from the drop down list and click on Add Selected Block.
[image]

The session log has no errors.  Did you get any error dialogs?

dsand47, 2023-02-14:
You have very long block names.  I suspect that you need to make the window wider to include the Add Selected Block button.

Mack9000, 2023-02-14:
[image]

Now that I see what I should be seeing.. It looks like this is about how it displays the pop up.. I can stretch out the view then I see what I'm looking for..   Thank you for the quick response.  

[image]

Mack9000, 2023-02-14:
Thank you.. This can be closed..

closed 2023-02-14 by Mack9000