JMRI/JMRI #11808: Add copy option to Timetable schedules

PUlvestad, 2023-02-18:
Feature request.
I have been playing with the Timetable tool for a couple days and today working with someone else on theirs.
It would very handy to be able to copy a schedule complete with trains for example when making a weekend schedule.

Peter

dsand47, 2023-02-18:
This sounds like a good idea.

A possible approach is to add a **Duplicate** action button to each record type.  When selected, a copy of the record and its children are created.  When the copy process is done, the new record will be selected.

PUlvestad, 2023-02-21:
Your approach is much better.

This is a great tool. I'm using it to do some of the documentation for my NMRA AP Chief Dispatcher certificate.

dsand47, 2023-03-02:
This has been implemented in PR #11825.

closed 2023-03-05 by danielb987