JMRI/JMRI #11836: Hide attribute changed on File Store

jerryg2003, 2023-03-07:
Using 5.3.3, Java 11.0.16.1, Windows 10.
Found this minor problem in a complex panel file but was able to recreate in a simple form.  The disturbing thing is that JMRI seems to be changing my panel file when I had just stored it and didn't make any further changes.

Simple example:
1. Create two CPE panels. Hide one by clicking the "x" in the upper right. Use File/Store All on PanelPro window to create an xml file (attached).
2. Immediately close JMRI by clicking Quit on PanelPro window.
3. Get pop-up box "do you want to Store All?"  [[Why did I get this when I had just done a Store All?] Click yes and save to a different panel file (attached as with "X" added to file name).
4. Compare two panel files: find difference in that "Hide" attribute for unhidden panel is "no" in original xml file and changed to "yes" in second file.

NOTE: had to add ".txt" to file names to get GitHub to upload them.

 
[TEST Panel Hide.xml.txt](https://github.com/JMRI/JMRI/files/10913837/TEST.Panel.Hide.xml.txt)
[TEST Panel Hide X.xml.txt](https://github.com/JMRI/JMRI/files/10913838/TEST.Panel.Hide.X.xml.txt)

dsand47, 2023-03-07:
Jerry,

This is an interesting timing issue.

PE and CPE include the visibility state when the store process runs.  When this is run from a menu, the panels may or may not be visible depending on the close box actions.  Normally they will be visible.

The store and compare process that checks for unsaved data runs during the shutdown process.  This occurs after the JmriJFRame panels have been set not visible.  The test store will then set those panels to hide=yes.  If a regular store had been done, the data will have hide=no which results in the dialog.

Now we get to the real problem.  The CPE and PE store processes use the panel visibility to set the "hide" attribute.  The load processes use the "hide" attribute to set the panel content visibility option, which has nothing to do with the panel visibility.  This has the inadvertent side effect of changing the content visibility option based on how the data was stored.

The solution is to change "ControlPanelEditorXml.java" and "PanelEditorXml.java" to use "showHidden()" to set the "hide" attribute during store instead of the panel visibility.

Dave Sand

jerryg2003, 2023-03-08:
`The CPE and PE store processes use the panel visibility to set the "hide" attribute. The load processes use the "hide" attribute to set the panel content visibility option, which has nothing to do with the panel visibility. This has the inadvertent side effect of changing the content visibility option based on how the data was stored. `

I appreciate the detailed review and explanation but confess I got lost somewhere in the middle of content visibility having nothing to do with panel visibility.  I hope it wasn’t too difficult to figure out.

closed 2023-03-09 by danielb987