JMRI/JMRI #11844: IP connection lost to DCC++ (CommandStation-EX)

prchal, 2023-03-09:
When there is short disconnection between CommandStation-EX + W5500 and RasPi4 - station reboots or eth switch reboots or cable disconnects - the connection doesn't go up again. Must restart JRMI (Faceless).
Log:
`2023-03-09 12:52:37,197 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2023-03-09 12:52:37,203 util.Log4JUtil                        INFO  - This log is appended to file: /etc/jmri/log/messages.log [main]
2023-03-09 12:52:37,204 util.Log4JUtil                        INFO  - This log is stored in file: /etc/jmri/log/session.log [main]
2023-03-09 12:52:37,729 apps.AppsBase                         INFO  - JmriFaceless version 5.3.3+R77156098f8 starts under Java 11.0.18 on Linux aarch64 v5.15.76-v8+ at Thu Mar 09 12:52:37 CET 2023 [main]
2023-03-09 12:52:38,108 apps.AppsBase                         INFO  - Starting with profile prchals_railroad.3d30ba30 [main]
2023-03-09 12:52:38,746 node.NodeIdentity                     INFO  - Using 248ac8e0-e427-494a-a82a-1891d0d2ea07 as the JMRI storage identity for profile id 3d30ba30 [AWT-EventQueue-0]
2023-03-09 12:52:39,012 jmrix.ConnectionConfigManager         INFO  - No local configuration found. [main]
2023-03-09 12:52:39,939 dccpp.DCCppInitializationManager      INFO  - DCC++ Initialization Complete with station type 'Unknown', version '0.0.0' and build 'Unknown' [main]
2023-03-09 12:52:40,476 swing.JmriJTablePersistenceManager    INFO  - Table preferences not found.
This is expected on the first time the "prchals railroad" profile is used on this computer. [main]
2023-03-09 12:52:41,270 dccpp.DCCppCommandStation             INFO  - changing maxNumSlots from 12 to 50 [AWT-EventQueue-0]
2023-03-09 12:52:41,509 dccpp.DCCppCommandStation             INFO  - Station Type set to 'DCC-EX' [AWT-EventQueue-0]
2023-03-09 12:52:41,510 dccpp.DCCppCommandStation             INFO  - Build set to 'devel-202302121935Z' [AWT-EventQueue-0]
2023-03-09 12:52:41,511 dccpp.DCCppCommandStation             INFO  - Version set to '4.2.18' [AWT-EventQueue-0]
2023-03-09 12:52:42,084 util.FileUtilSupport                  INFO  - File path program: is /opt/JMRI/ [main]
2023-03-09 12:52:42,085 util.FileUtilSupport                  INFO  - File path preference: is /etc/jmri/prchals_railroad.jmri/ [main]
2023-03-09 12:52:42,090 util.FileUtilSupport                  INFO  - File path profile: is /etc/jmri/prchals_railroad.jmri/ [main]
2023-03-09 12:52:42,091 util.FileUtilSupport                  INFO  - File path settings: is /etc/jmri/ [main]
2023-03-09 12:52:42,092 util.FileUtilSupport                  INFO  - File path home: is /root/ [main]
2023-03-09 12:52:42,092 util.FileUtilSupport                  INFO  - File path scripts: is /opt/JMRI/jython/ [main]
2023-03-09 12:52:42,575 withrottle.FacelessServer             INFO  - Published ZeroConf service for 'prchals railroad._withrottle._tcp.local.' on fe80:0:0:0:2f34:a5c4:737c:643c%eth0:12090 [WiThrottleServer]
2023-03-09 12:52:42,576 withrottle.FacelessServer             INFO  - Published ZeroConf service for 'prchals railroad._withrottle._tcp.local.' on 172.25.160.177:12090 [WiThrottleServer]
2023-03-09 12:52:42,577 withrottle.FacelessServer             INFO  - Creating new WiThrottle DeviceServer(socket) on port 12090, waiting for incoming connection... [WiThrottleServer]
2023-03-09 12:52:43,384 server.WebServer                      INFO  - Starting Web Server on port 12080 [WebServer]
2023-03-09 12:52:44,315 server.WebServer                      INFO  - Starting ZeroConfService _http._tcp.local for Web Server with properties {path=/, json=5.4.0} [WebServer]
2023-03-09 13:02:37,388 withrottle.FacelessServer             INFO  - Creating new WiThrottle DeviceServer(socket) on port 12090, waiting for incoming connection... [WiThrottleServer]
2023-03-09 13:02:37,392 withrottle.DeviceServer               INFO  - Received Name: Engine Driver 6350 [DeviceServer-1]
2023-03-09 13:04:09,280 jmrix.AbstractMRTrafficController     ERROR - run: Exception: java.net.SocketException: Connection reset in class jmri.jmrix.dccpp.network.DCCppEthernetPacketizer [network.DCCppEthernetPacketizer Receive thread]
java.net.SocketException: Connection reset
	at java.base/java.net.SocketInputStream.read(SocketInputStream.java:186)
	at java.base/java.net.SocketInputStream.read(SocketInputStream.java:140)
	at java.base/java.io.DataInputStream.read(DataInputStream.java:149)
	at jmri.jmrix.AbstractMRTrafficController.readByteProtected(AbstractMRTrafficController.java:959)
	at jmri.jmrix.dccpp.DCCppPacketizer.loadChars(DCCppPacketizer.java:152)
	at jmri.jmrix.AbstractMRTrafficController.handleOneIncomingReply(AbstractMRTrafficController.java:1063)
	at jmri.jmrix.AbstractMRTrafficController.receiveLoop(AbstractMRTrafficController.java:880)
	at jmri.jmrix.AbstractMRTrafficController$2.run(AbstractMRTrafficController.java:803)
	at java.base/java.lang.Thread.run(Thread.java:829)
2023-03-09 13:04:09,296 jmrix.AbstractMRTrafficController     WARN  - Exception java net:  [network.DCCppEthernetPacketizer Receive thread]
java.net.SocketException: Connection reset
	at java.base/java.net.SocketInputStream.read(SocketInputStream.java:186)
	at java.base/java.net.SocketInputStream.read(SocketInputStream.java:140)
	at java.base/java.io.DataInputStream.read(DataInputStream.java:149)
	at jmri.jmrix.AbstractMRTrafficController.readByteProtected(AbstractMRTrafficController.java:959)
	at jmri.jmrix.dccpp.DCCppPacketizer.loadChars(DCCppPacketizer.java:152)
	at jmri.jmrix.AbstractMRTrafficController.handleOneIncomingReply(AbstractMRTrafficController.java:1063)
	at jmri.jmrix.AbstractMRTrafficController.receiveLoop(AbstractMRTrafficController.java:880)
	at jmri.jmrix.AbstractMRTrafficController$2.run(AbstractMRTrafficController.java:803)
	at java.base/java.lang.Thread.run(Thread.java:829)
2023-03-09 13:04:09,299 jmrix.AbstractMRTrafficController     ERROR - Exit from rcv loop in class jmri.jmrix.dccpp.network.DCCppEthernetPacketizer [network.DCCppEthernetPacketizer Receive thread]
2023-03-09 13:04:10,309 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [Connection Recovery 172.25.160.92:2560]
java.lang.IllegalStateException: Task already scheduled or cancelled
	at java.base/java.util.Timer.sched(Timer.java:402)
	at java.base/java.util.Timer.schedule(Timer.java:249)
	at jmri.util.TimerUtil.schedule(TimerUtil.java:51)
	at jmri.jmrix.dccpp.network.DCCppEthernetAdapter.keepAliveTimer(DCCppEthernetAdapter.java:100)
	at jmri.jmrix.dccpp.network.DCCppEthernetAdapter.connect(DCCppEthernetAdapter.java:46)
	at jmri.jmrix.AbstractNetworkPortController.reconnectFromLoop(AbstractNetworkPortController.java:305)
	at jmri.jmrix.AbstractPortController$ReconnectWait.run(AbstractPortController.java:458)
	at java.base/java.lang.Thread.run(Thread.java:829)
2023-03-09 13:04:14,277 jmrix.AbstractMRTrafficController     WARN  - Timeout on reply to message: 0 consecutive timeouts = 0 in network.DCCppEthernetPacketizer [network.DCCppEthernetPacketizer Transmit thread]
2023-03-09 13:06:03,604 withrottle.FacelessServer             INFO  - Creating new WiThrottle DeviceServer(socket) on port 12090, waiting for incoming connection... [WiThrottleServer]
2023-03-09 13:06:03,605 withrottle.DeviceServer               INFO  - Received Name: Engine Driver 6350 [DeviceServer-2]
`

bobjacobsen, 2023-03-09:
(edited original post for better formatting)

prchal, 2023-03-09:
Maybe it would be better to end the proces since it can be restarted it via systemd.

closed 2023-05-09 by stale[bot]