JMRI/JMRI #11846: Connection to DCS210+ via Decorder Pro or Panel Pro has yet to work

jimpriceeverett, 2023-03-09:
Been troubleshooting this issue for a week now. 

I have been attempting to just do a clean install of 5.2 as Upgrade from 4.26 to 5.2 failed.

As I suspect that recent changes with how OneDrive integration with Windows 11 appears to have caused issues with the resolution of path names.

I am resigned with rebuilding of tables as it appears that the Switch data may not be there, I anticipated that as they were DS54s from earlier in this century... PM42 & BDL168s seem to be powering track as I can run locomotives as the ones that I had programed with addresses still respond.  

I wouldn't be posting via github vs groups.io ....but since I had already started digging the rabbit hole...

[systemInfo.txt](https://github.com/JMRI/JMRI/files/10932138/systemInfo.txt)
[session.log](https://github.com/JMRI/JMRI/files/10932144/session.log)
[messages.log](https://github.com/JMRI/JMRI/files/10932145/messages.log)
[messages captured during initial  Upgrade.log](https://github.com/JMRI/JMRI/files/10932350/messages.captured.during.initial.Upgrade.log)

KenC57, 2023-03-09:
I suggest confirming that COM3 is correct. Use the Windows Device Manager to view the "Ports (COM & LPT)" entry. As you plug and unplug the connection to the DCS, the COM port should show up and leave the list.
Next, right click on it and see that the status says ok.
Then recheck what Digitrax says about installing the drivers.
[https://www.digitrax.com/tsd/product/PR4/](https://www.digitrax.com/tsd/product/PR4/)
Review page 28 of the Evolution manual. Does it light up as described?
Also open the LocoNet Monitor (LocoNet->Monitor) and then watch for traffic. Things like throttle traffic, detection messages, etc....
Report how it goes.

jimpriceeverett, 2023-03-09:
Thanks for the direction,,,COM3 does not seem to be the issue...but it pointed me to the right page above the DCS210+ beeps paragraph...I am in the process of evaluating the wiring as I did entice a beep or two fiddling with the Type B end of the USB cable.
I will let you all know once I rework the wiring...

jimpriceeverett, 2023-03-10:
Ok, we are in better shape now...not sure why PM42 didn't like the SE8C with external power...continued complaining about COM3...went back to setup of command station and system as if it was brand new...eventually it took a few passes on the New Loco - Read type from decoder...backing out to the Getting Started...Quitting Decoder Pro until the COM3 Port appeared in Device Manager...and then repeating the New Loco - Read type from Decoder process until it worked...
Now the fun part...rebuilding roster...

closed 2023-03-10 by jimpriceeverett