JMRI/JMRI #11847: IdTag Date load / store xml format

icklesteve, 2023-03-10:
IdTags currently load / store the dateLastSeenField Date field in xml as 
`DateFormat.getDateTimeInstance(DateFormat.MEDIUM, DateFormat.MEDIUM).format(date)`

which results in xml such as

```
<idtag>
      <systemName>ID0A0A021C1B</systemName>
      <whereLastSeen>MR78</whereLastSeen>
      <whenLastSeen>Mar 9, 2023, 6:14:54 AM</whenLastSeen>
</idtag>
```

The issues with this are:

Millisecond value is not restored.
When a rake of individually tagged trucks passes over a reader, this is required to later ascertain the order.
More than 1 truck may have passed over a reader within 1 second.

When a user changes the display language settings, the format changes and all previous values are lost.
En-US to En-GB > ( `Mar 9, 2023, 4:15:34 AM` to `9 Mar 2023, 06:24:10` )

I propose to change whenLastSeen to the ISO 8601 format, currently in use with RosterEntry last modified xml, eg.
`2023-03-09T06:34:55.472+00:00`

Currently putting some unit tests behind this.

danielb987, 2023-03-10:
What happens with old files if the format is changed? Will the old files still be loadable? Can an old version of JMRI read the new format?

danielb987, 2023-03-10:
What happens if you do:

```
<idtag>
      <systemName>ID0A0A021C1B</systemName>
      <whereLastSeen>MR78</whereLastSeen>
      <whenLastSeen>2023-03-09T06:14:54.472+00:00</whenLastSeen>
</idtag>
```
Maybe the xml loader can handle that.

PR #11865 (2023-03-20): update DefaultIdTag xml date format
Updates xml date format to ISO 8601
fixes #11847 

adds more DefaultIdTag unit tests
white space formatting
closed 2023-03-20 by danielb987