JMRI/JMRI #11853: CV Description

Blueman-2, 2023-03-12:
Issue Type: __Feature Request__

Please can we have an extra column on the CV's tab containing the description of the CV.

We already have tooltips on the other pages showing the CV number when hovering over the input box and I feel that this would be a good enhancement making the screen more user friendly.

PanelPro 5.2+R760b98537f
Java HotSpot(TM) 64-Bit Server VM Oracle Corporation 11.0.15+8-LTS-149 mixed mode
Windows 10 10.0 amd64

Blueman-2, 2023-03-12:
Sorry just found and is duplicated eleswhere,

danielb987, 2023-03-12:
@bobjacobsen 
What's your view about this? I know little about CVs, but my guess is that tooltips might refer to only a part of a CV and that there might be several tooltips for the same CV, and that a single tooltip might refer to several CVs.

bobjacobsen, 2023-03-12:
CV29 is an example of a single CV that has multiple (5 or 6) different contents.  CV17-18 is an example of a single variable that's across two different CVs.  There are more complicated examples, too.

The color coding takes this into account, so when you change a variable, all the CVs it touches have their color changed.  But that's usually not reversible. It'd hard to _automatically_ create something that's generally useful.  

You could do something for the simple cases of single content in a single CV.  That might still be helpful.  Then the rest would be labeled with something like "multiple contents" or similar.

We should probably explore a bit more what's underlying the request.  @Blueman-2, when would this be useful to you?  Perhaps we can find another way to achieve that goal.

Blueman-2, 2023-03-13:
Hi there

Thank you for the reply.

My idea was just to have an extra column showing a simple description. e.g.

CV, Value, Description,.....
5,128, Maximum speed
19, 0, Consist address
81, 68, Free speed curve
Etc.

I know that there are CV's which are more complex, 17, 18 and 29 for starters but you could have a "Refer to manual" on that type.

The reasoning behind the request is that decoders have become more complex and some, Zimo have a one size fits all approach with the definition in JMRI, so I have 34 tabs to go through if I want to find what a CV change relates to, which can be frustrating.

Regards

Paul

uli-2016, 2023-04-17:
I think Paul would like to see the description of each CV in the table. I think the idea is good.

[image]

bobjacobsen, 2023-04-18:
PR #11958 adds the basic capability here.

PR #11958 (2023-04-19): Add "where used" tooltips to CV pane
See Issue #11853 for background.

CVs that appear in multiple variables will show a semicolon separated list of the variables in no particular order.

Closes #11853
closed 2023-04-19 by danielb987