JMRI/JMRI #11862: Uhlenbrock Daisy II Zentrale with USB 63120 Interface Fails

johnmrichey, 2023-03-18:
Issue Type: __Bug__

I have the Daisy II Zentrale with the USB 63120 Interface.  When adding a Loco it fails with error 306.

DecoderPro 5.2+R760b98537f
OpenJDK 64-Bit Server VM Azul Systems, Inc. 11.0.16.1+1-LTS mixed mode
Windows 10 10.0 amd64

2023-03-17 19:26:33,364 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2023-03-17 19:26:33,380 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\johnm\JMRI\log\messages.log [main]
2023-03-17 19:26:33,380 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\johnm\JMRI\log\session.log [main]
2023-03-17 19:26:33,458 apps.AppsBase                         INFO  - DecoderPro version 5.2+R760b98537f starts under Java 11.0.16.1 on Windows 10 amd64 v10.0 at Fri Mar 17 19:26:33 PDT 2023 [main]
2023-03-17 19:26:33,940 gui3.Apps3                            INFO  - Starting with profile My_JMRI_Railroad.3f2b5bcd [main]
2023-03-17 19:26:34,127 node.NodeIdentity                     INFO  - Using 8b20528a-1d85-46ef-beb4-9a7c75c6ae57 as the JMRI storage identity for profile id 3f2b5bcd [AWT-EventQueue-0]
2023-03-17 19:26:34,330 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "LocoNet" [main]
2023-03-17 19:26:34,455 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "LocoNet" [main]
2023-03-17 19:26:34,665 locobuffer.LocoBufferAdapter          INFO  - LocoBuffer (serial) adapter set hardware flow control, mode=2 RTSCTS_OUT=2 RTSCTS_IN=1 [main]
2023-03-17 19:26:34,665 locobuffer.LocoBufferAdapter          INFO  - COM1 port opened at 115200 baud with DTR: true RTS: true DSR: false CTS: false  CD: false [main]
2023-03-17 19:26:34,806 loconet.LnPacketizer                  INFO  - lnPacketizer Started [main]
2023-03-17 19:26:34,875 managers.ManagerDefaultSelector       WARN  - The configured LocoNet for interface jmri.ThrottleManager can not be found so will use the default LocoNet2 [main]
2023-03-17 19:26:34,875 managers.ManagerDefaultSelector       WARN  - The configured LocoNet for interface jmri.PowerManager can not be found so will use the default LocoNet2 [main]
2023-03-17 19:26:35,641 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2023-03-17 19:26:35,641 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\johnm\JMRI\My_JMRI_Railroad.jmri\ [main]
2023-03-17 19:26:35,641 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\johnm\JMRI\My_JMRI_Railroad.jmri\ [main]
2023-03-17 19:26:35,641 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\johnm\JMRI\ [main]
2023-03-17 19:26:35,641 util.FileUtilSupport                  INFO  - File path home: is C:\Users\johnm\ [main]
2023-03-17 19:26:35,641 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]
2023-03-17 19:27:11,613 jmrit.AbstractIdentify                WARN  - Timeout talking to command station (306) readng CV 8, trying Paged mode [AWT-EventQueue-0]
2023-03-17 19:27:35,736 jmrit.AbstractIdentify                WARN  - Restoring Direct Byte mode [AWT-EventQueue-0]
2023-03-17 19:27:35,736 jmrit.AbstractIdentify                WARN  - Stopping due to error: Timeout talking to command station (306) [AWT-EventQueue-0]
2023-03-17 19:34:03,103 loconet.LnPacketizer                  WARN  - LocoNet message with opCode: c0 ended early. Expected length: 6 seen length: 3 unexpected byte: ed [LocoNet receive handler]
2023-03-17 19:34:11,108 jmrit.AbstractIdentify                WARN  - Timeout talking to command station (306) readng CV 8, trying Paged mode [AWT-EventQueue-0]
2023-03-17 19:34:35,236 jmrit.AbstractIdentify                WARN  - Restoring Direct Byte mode [AWT-EventQueue-0]
2023-03-17 19:34:35,236 jmrit.AbstractIdentify                WARN  - Stopping due to error: Timeout talking to command station (306) [AWT-EventQueue-0]

Pugwash1, 2023-03-18:
> 2023-03-17 19:26:34,875 managers.ManagerDefaultSelector WARN - The configured LocoNet for interface jmri.ThrottleManager can not be found so will use the default LocoNet2 [main]
2023-03-17 19:26:34,875 managers.ManagerDefaultSelector WARN - The configured LocoNet for interface jmri.PowerManager can not be found so will use the default LocoNet2 [main]

John
It looks like you have 2 Interfaces configured. 
Top right of the decodepro screen there is a "Program On" combo box, does it say "Loconet2" or "Loconet", it should be lconet2 as the other one appears to be balked.

johnmrichey, 2023-03-18:
Steve,The first LocoNet was messed up so I created the second and disabled the first, but then the second stopped working as well, same errors.Where is the preferences files on Windows so I can start anew please?  On Mac I know where it is.Thanks

johnmrichey, 2023-03-18:
Steve,Which settings should I use for the Zentrale please?I have the 63120 as the USB interface.ThanksJohn R.

Pugwash1, 2023-03-19:
How to manage your profiles including adding a new one is covered here [https://www.jmri.org/help/en/html/setup/profiles.shtml](https://www.jmri.org/help/en/html/setup/profiles.shtml) 
As to which options you need I do not know as I do not have your hardware.

johnmrichey, 2023-03-19:
Steve,Thanks for the info.I had reported on this before about a year ago, and thought it might be resolved by now.  Need to dig up that old thread.ThanksJohn R.

Pugwash1, 2023-03-25:
@johnmrichey  if this thread on groups.io [https://groups.io/g/jmriusers/topic/69677332#168774](https://groups.io/g/jmriusers/topic/69677332#168774)  is the problem then a quick work around would be to use a jython script to inject the missing LACK message.

closed 2023-05-31 by stale[bot]